FabMo/FabMo-Engine #67: Tool Doesn't respond anymore to Job / Commands

antoinedelhaye, 2015-01-20:
Hey,
Just get back from holidays.
I updated the Engine to the last version.
I can use the keypad normally.
It is not possible anymore to 0 an axis and when I send a job, its status goes straight to 100% and the tool doesn't move anymore.

Any idea ?

jlucidar, 2015-01-21:
You delete the zero buttons in commit 2d3c8e3e14d8ae6c24b5b319447931c07e79c6a7 .
I will fix it .
The keypad could be my fault as I changed the code with my tries for the fixed mode.
It could be integrated directly in the Engine jog function, what do you think @ryansturmer  ? I can work on it.

jlucidar, 2015-01-21:
I got a problem with the keypad too. here is the log where it fails : 
```
debug: machine jog [machine]
info: Starting jog in x direction. [manual]
info: Got a machine state change: manual [machine]
info: Keeping jog alive. [g2]
info: Stopping jog. [manual]
debug: stopJog() [g2]
debug: Not handling this queue report (i:0 o:1 r:19) because pause or quit pendi
ng [g2]
debug: Not handling this queue report (i:0 o:1 r:20) because pause or quit pendi
ng [g2]
debug: Not handling this queue report (i:0 o:1 r:21) because pause or quit pendi
ng [g2]
debug: Not handling this queue report (i:0 o:1 r:22) because pause or quit pendi
ng [g2]
debug: Not handling this queue report (i:0 o:1 r:23) because pause or quit pendi
ng [g2]
debug: Not handling this queue report (i:0 o:1 r:24) because pause or quit pendi
ng [g2]
debug: Not handling this queue report (i:0 o:1 r:25) because pause or quit pendi
ng [g2]
debug: Not handling this queue report (i:0 o:1 r:26) because pause or quit pendi
ng [g2]
debug: Not handling this queue report (i:0 o:1 r:27) because pause or quit pendi
ng [g2]
debug: HONOR this status report because it doesn't contain hold data or because
hold=0 [g2]
debug: Manual runtime got a state change from 5 to 3 [manual]
info: Got a machine state change: idle [machine]
debug: Not handling this queue report (i:0 o:1 r:28) because pause or quit pendi
ng [g2]
debug: Clearing the queue by hack (closing and reopening the port) [g2]
debug: closing [g2]
debug: opening again [g2]
debug: Queue cleared. [g2]
```

The machine is stuck after its restart. I'm still using my old version of G2 though ...

ryansturmer, 2015-01-21:
This is due to a shortfall in the G2 firmware, which doesn't process a queue clear properly.  The latest version of g2.js in the dashboard is meant to interact with the latest version of G2 firmware, which attempts to fix these issues.  I hope to have this version complete before I come into shopbot on thursday.  For now, you can use an older version of dashboard-dual if you still need keyboard operation, but feedhold/flush (which is needed for kb operation) is broken.

ryansturmer, 2015-01-21:
We didn't see this before because the queue flush was not actually flushing, now that we're expecting it to actually work, it's creating problems, since it doesn't work as advertised in G2.  I have a candidate fix for it.

ryansturmer, 2015-02-21:
This is fixed in the latest master/g2 firmware. Hold/flush are working again, and the normal job commands are in place and in dashboard.js

closed 2015-02-21 by ryansturmer