FabMo/FabMo-Engine #69: Normalizing of tool API

ryansturmer, 2015-01-22:
As I document the API for the engine, I'm finding a lot of inconsistencies and weird behaviors, and we do not respond in a normalized fashion.

I propose that we adopt a standardized response format, and I like the look of this.  It's just a guideline, but is sensible and lightweight:

http://labs.omniti.com/labs/jsend

jlucidar, 2015-01-22:
I like it too. let me know if you need me for the changes.

ryansturmer, 2015-01-23:
What do you think of that jsend convention?

ryansturmer, 2015-01-23:
I welcome your help!  Do it in a feature branch though, since this will bubble up changes to the dashboard, and we're looking to test a release candidate today.  If you don't get around to it though, I'll start working on it soon.

jlucidar, 2015-01-26:
I will work on it this week.

jlucidar, 2015-02-16:
The issue is almost solved, I just need some help on the backbone side.
Do you have any idea of how to change this file to become jsend compliant ?
https://github.com/ShopBotTools/FabMo-Engine/blob/2ef6865fb7d2f940a007b7804d0f99fc54be7ef7/dashboard/static/js/models.js

The branch might need more tests before merging.

ryansturmer, 2015-02-17:
I will merge and test this today.  Don't worry about the backbone stuff, I will be changing some of this in the near future, and I can clean it up.  Thanks for getting this done.

jlucidar, 2015-02-17:
Well, the problem is that you can't launch apps anymore, perhaps considering a rollback on the dashboard routes to the old protocol is the solution. Another solution might be here : https://github.com/johntimothybailey/backbone.jsend . let me know where I can help next !

ryansturmer, 2015-02-17:
I'll take a look today, I'm sure it's straightforward.  The backbone stuff in the dashboard needs some cleanup anyway.

ryansturmer, 2015-02-18:
Hey, this is all pretty straightforward.  I merged into the master before I should have, there are still some bugs here, but nothing show-stopping.  The fix for backbone is a one-liner, you just have to override the parse() method and have it fish out the model data when it does a sync.  There are other issues though, but I should have those resolved shortly.  Mostly it's issues inside apps, like the job manager.

ryansturmer, 2015-02-21:
I'm calling this one closed.  There's a few glitches here and there, but the API is normalized, even if the apps haven't all caught up yet.  Good work Jimmy.

closed 2015-02-21 by ryansturmer