FabMo/FabMo-Engine #75: Report from running hood files

tedh-shopbot, 2015-03-10:
Ran production files for hood several times ... mixed results. Cutting good and aggressive ramping seems fine. But had several lockups that required reboot. Once with handwheel, once trying to upload file ... after this one, history of jobs disappeared. Reboot brought it back. Most concerning was that one file locked up just after router turned on ... electrical noise (or mishandling of dwell)?

gordonab, 2015-03-11:
I also had an issue with the hand wheel locking up this afternoon. Also, had an issue with a move not executing using the new editor and the report let me know that I needed a feedrate (F word) in the G1 I was trying to execute. I like this new feature.

ryansturmer, 2015-03-11:
Ideally, errors like the one about the feedrate should propagate "properly" to the dashboard, and give a user friendly error report.  The debug output is just an interim step.

Handwheel related lock-ups should be improved by the feedhold fixes that are coming soon.

When you say the history of jobs "disappeared" do you mean that while you were in the jobs manager page, the jobs actually went away?  or you refreshed the jobs page, and on refresh they were gone?

When the file locked up right after the router turned on - did you recover the debug log?  Or was the tool unreachable via the network at that point?  In some of these cases, I'm starting to suspect that things like electrical noise are actually putting the wifi in a bad state, and connection to the tool is being lost.  The websockets version of the status reports should at least make such failures (if they exist) easier to resolve.

tedh-shopbot, 2015-03-11:
I'm still running files and doing production. Look forward to seeing how the websocket works and the improved feed hold. At the moment, manual zeroing is making me pretty inefficient BUT it is sure nifty to walk away from the tool and be able to monitor what it is up to. 

*I'm thinking that the Dashboard wants a page or app for "Frequently used Files" for saving macros like zeroing, etc. -- something that offers a more efficient display than the main apps page ... the graphics are nice, but maybe just a list, a label and a color or simple icon ... because some people may have a lot of them. We are definitely going to need ways to organize pages in the Dashboard ...

ryansturmer, 2015-03-13:
Closing this, opening three new issues in its place: #78 #79 and #80

closed 2015-03-13 by ryansturmer