FabMo/FabMo-Engine #77: Dashboard features ...

tedh-shopbot, 2015-03-11:
*I'm thinking that the Dashboard wants a page or app for "Frequently used Files" for saving macros like zeroing, etc. -- something that offers a more efficient display than the main apps page ... the graphics are nice, but maybe just a list, a label and a color or simple icon ... because some people may have a lot of them. We are definitely going to need ways to organize pages in the Dashboard ...

tedh-shopbot, 2015-03-11:
ok ... I cut the above out and made new issue ... =t

ryansturmer, 2015-03-13:
Per our discussion today, the these "macros" will look a lot like what we now call jobs, except for a few key differences:
* They will point to files that are mutable - ie, they can be edited in place (or from the perspective of the user they will be, even if they underlying files aren't)
* They will "skip" the job queue and run directly on the tool - this makes them more like callable routines
* They will map _somehow_ to C# commands in shopbot, or M# commands in G-Code such that things like tool changes etc. can be managed.

jlucidar, 2015-10-11:
Just to make a little point on the situation there : 
Points 1 & 2 are implemented, I can't get the third case working. 
What I tried to do : 
- Creating a "C3" macro with a single line of OpenSBP code
- creating a "test" macro that call C3 after a few lines of OpenSBP.
=> the "test" macro tell me that C3 doesn't exist.

ryansturmer, 2015-10-13:
Jimmy can you open this up as another issue with specific steps to reproduce?

closed 2015-10-13 by ryansturmer