FabMo/FabMo-Engine #80: Propagation of G-Code/OpenSBP errors to dashboard

ryansturmer, 2015-03-13:
G codes that are bogus need to be reported to the dashboard in a way that is NOT the debug log.  If a file bombs because of a parse error, or a programming error of some other variety, the nature of that failure (including information such as the line number, etc) needs to be propagated back to the user in a way that allows them to go back and fix their jobs.

ryansturmer, 2015-06-24:
This has been resolved for OpenSBP, but doesn't currently work for G-Code

RobMackie, 2021-03-24:
Ok. this is a bit weird. I ran the following gcode successfully, with no errors reported and the machine did just what was instructed if   you pretend the garbage line isn't there
G90
G20
F60
G1 X4
Any kind of error
G1 X0
M30

So I also tried this one:
G90
G20
F60
G1 X4
G1 Q400
G1 X0
M30
Handled file like it was a good file. No problem, ran all the gcode that is legit and ignored the erroneous line.
And this one
G90
G20
F60
G1 X4
G1 
G1 X0
M30

and this one:
G90
G20
F60
G1 X4
G0
G1 X0
M30

I'll collect some log files and add them here to see where we're handling these lines. Is this the behavior we want? zero error reporting to the user on a bad input file? it might be "code" that has an effect on another machine.

closed 2022-10-27 by RobMackie