FabMo/FabMo-Engine #81: A,B,C axis display on dashboard

ryansturmer, 2015-03-13:
A,B,C axes are not currently displayed on the dashboard.  These should be displayed, but only if they are currently available.  A tool with only 3 axes should show 3 axes, I think, but one with 6 should display them all.  Perhaps the axes displayed should be selectable (so if you had a 4 axis capable tool but only had 3 axes installed, you wouldn't be forced to see an A axis all the time.)

tedh-shopbot, 2015-03-13:
Perhaps the latter would be the most elegant. But on SB3 at the moment, it's just determined by the user specifying how many. I'm thinking the tool does not know how many are hooked up, or how many for a given work session the user wants to display. So ... I've previously set it up that the user selects what to display, and if a file requires action in an axis that is not displayed/enabled, then when it is run it stops and prompts the user to enable the additional axes.

ryansturmer, 2016-02-03:
Done.

closed 2016-02-03 by ryansturmer