FabMo/FabMo-Engine #89: Gcode circles not executing - G3 command works, but not G2 command

gordonab, 2015-04-02:
The Gcode circle/arc command G2 is not executing, yet G3 is. This command will execute: G3X4Y4I0J2F60 It goes counterclockwise in what appears to be a 4 inch diameter circle.

This same command: G2X4Y4I0J2F60, except for being a G2 (clockwise circle) doesn't execute. There are 2 error messages in the log after each attempt to run a G2 command:

    error: 201,STAT_MINIMUM_LENGTH_MOVE,move is less than minimum length [machine]
    error: 155,STAT_ARC_SPECIFICATION_ERROR,generic arc specification error [machine]

Further testing:

This is only an issue when the Start and End points are the same, as is the case when specifying a full circle. Arcs seem to be working well, but will continue to test for any further issues.

aldenhart, 2015-04-03:
When specifying a full circle the end points are omitted - according to the spec. If they are the same it’s an error. Not to say there is still not an issue.  Try G2J2F60

A

gordonab, 2015-04-03:
I looked this up in Kramer's version of the NIST spec (The NIST RS274NGC Interpreter - Version 3) on page 24 it talks about center point circles:

****************************************************************************************************
3.5.3.2 Center Format Arc
.
.
.
When the XY-plane is selected, program G2 X… Y… Z… A… B… C… I… J… (or use G3
instead of G2). The axis words are all optional except that at least one of X and Y must be used. I
and J are the offsets from the current location (in the X and Y directions, respectively) of the
center of the circle. I and J are optional except that at least one of the two must be used. It is an
error if:
• X and Y are both omitted,
• I and J are both omitted.
.
.
.
****************************************************************************************************

Up until the last version of the G2 core this worked for both CW and CCW circles and arcs. And, including the X & Y works for CCW (G3) circles, but not CW (G2) circles.

Is there another section in the NIST spec that adds additional information on these?

aldenhart, 2015-04-03:
> WI looked this up in Kramer's version of the NIST spec (The NIST RS274NGC Interpreter - Version 3) on page 24 it talks about center point circles:

We use the LinuxCNC specification as the “addendum” to Kramer. Here’s what it says about arcs:

http://linuxcnc.org/docs/html/gcode/gcode.html#sec:G2-G3-Arc <http://linuxcnc.org/docs/html/gcode/gcode.html#sec:G2-G3-Arc>

Without copying al of it, here’s some cherry-picked parts from 5.1 - Center Format Arcs:

"The end point of the arc along with the offset to the center of the arc from the current location are used to program arcs that are less than a full circle. It is OK if the end point of the arc is the same as the current location.”

and "No axis words and one or more offsets must be programmed for full circles.”

I read that to mean that only an arc with no X and no Y (G17 plane) should be treated as a full circle. And that an arc that DOES have X and/or Y specified and is the same as the current position is OK, but should be treated as a zero length move. This is safer than assuming the user wanted a 360 degree arc, which could ruin the job, crash the tool, or worse.

So returning a Zero-Length-Move exception is not an error, just an advisory. I am not sure where the arc specification error comes from, but usually that’s the case when the offsets provided don’t specify a rational center, and the arc is incomputable.

I can’t run this as I’m way, way overseas right now, but hopefully this helps.

Alden

ryansturmer, 2016-02-02:
@gordonab We have ruled some of the other circle issues as problems with node-serialport.  What about this one?

closed 2016-05-19 by ryansturmer