FabMo/FabMo-Engine #100: Problems after accidentally running metric file in inches ...

tedh-shopbot, 2015-04-18:
Accidentally ran a metric file from Jonathan's thing ...

Took off to china, so had to stop with STOP. On restarting things, seem connected and communicating, etc. Z works. But X and Y totally dead for handwheel and move commands (not even changes in DRO, though Z works). Tried various resets and reinstalled V9 firmware. Same -- only Z works ... Using VD to set to inches seemed w/o effect ...

It's as if going to metric reset something I am unable to over-ride with new install ... can't be in firmware (?) so ...

tedh-shopbot, 2015-04-19:
Seems like one issue after this event is that all the configuration data is gone, at least as displayed by the configuration page? Is there a method to repopulate it?

ryansturmer, 2015-04-19:
Please post the contents of the log on startup

ryansturmer, 2015-04-20:
Also if you have a copy of the file that was run, or instructions to reproduce the effect in his app, that would be helpful.

aldenhart, 2015-04-20:
Are you guys running current edge? What’s the build number? Thanks

tedh-shopbot, 2015-04-20:
Ryan ... forgot to post data when at ShopBot last night. Will do this am ...
Alden ... ditto ... but definitely close to current edge. But I'm not thinking this one is g2 ...
BTW, finally getting around to playing with V9 (esp and TI drivers) on Handibot ...

aldenhart, 2015-04-20:
Let me know what you find. They should have enough oomph to drive it, but we'll see.

tedh-shopbot, 2015-04-21:
Here's log data when I just fire up. Used handwheel to move Z; X and Y both failed; no values filled in for any of the configuration settings. G2=edge, last merged 4/9/15

info: Checking engine data directory tree... [engine]
info: Loading engine configuration... [engine]
info: Applying engine configuration... [engine]
info: Configuring database... [engine]
debug: Saving config to /opt/fabmo/config/engine.json [config]
info: Databases are clean. [files]
info: Cleaning up database... [engine]
info: Connecting to G2... [engine]
info: Opening control port /dev/ttyACM0 [g2]
info: Dual USB since control port and gcode port are different. (/dev/ttyACM0,/dev/ttyACM1) [g2]
info: Configuring G2... [engine]
debug: Configuring driver... [g2_config]
info: Getting G2 firmware version... [engine]
info: G2 Firmware Build: 83.05 [engine]
info: Loading OpenSBP Commands... [engine]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/Interpolate.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/custom.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/cut.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/experimental.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/file.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/jog.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/leveler.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/move.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/setting.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/value.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/zero.js" [opensbp]
info: Configuring OpenSBP runtime... [engine]
info: Configuring dashboard... [engine]
info: Loading dashboard apps... [engine]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/app_manager" because it already exists. [app_manager]
info: Loading macros... [engine]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/configuration" because it already exists. [app_manager]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/editor" because it already exists. [app_manager]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/handwheel" because it already exists. [app_manager]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/job_manager" because it already exists. [app_manager]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/macro_manager" because it already exists. [app_manager]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/network_manager" because it already exists. [app_manager]
debug: Not decompressing app "/opt/fabmo/apps/77b248c0-c67a-11e4-949d-dd104c32afbf.fma" because it already exists. [app_manager]
debug: Not decompressing app "/opt/fabmo/apps/8075f460-c388-11e4-9836-7b1688251980.fma" because it already exists. [app_manager]
debug: Not decompressing app "/opt/fabmo/apps/89416390-c388-11e4-9836-7b1688251980.fma" because it already exists. [app_manager]
debug: Not decompressing app "/opt/fabmo/apps/8d83a3e0-c677-11e4-949d-dd104c32afbf.fma" because it already exists. [app_manager]
info: Setting up the webserver... [engine]
info: Configuring cross-origin requests... [engine]
info: Cofiguring upload directory... [engine]
info: Loading routes... [engine]
debug:   Loaded routes from "/opt/fabmo/app/routes/config.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/dashboard.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/direct.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/file_commands.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/jobs.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/kill.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/log.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/macros.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/network_manager.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/passthrough.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/websocket.js" [routes]
info: FabMo Engine listening at http://0.0.0.0:9876 [engine]
debug: client connected [websocket]
info: Got a machine state change: manual [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
[... 194 lines cut ...]
[... cut ...]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]

tedh-shopbot, 2015-04-21:
Here's the output from Putty after I did what I thought was a re-install and restart of FabMo:

info: FabMo Engine listening at http://0.0.0.0:9876 [engine]

debug: client connected [websocket]
debug: client disconnected [websocket]
debug: client connected [websocket]
info: Got a machine state change: manual [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because mach                                                   ine is alarmed [machine]
[... cut ...]
info: Got a machine state change: idle [machine]
info: Connected ShopBot runtime. [sbp]
UNCAUGHT EXCEPTION
TypeError: Cannot call method 'apply' of undefined
    at Object.immediate._onImmediate (timers.js:363:16)
    at processImmediate [as _immediateCallback] (timers.js:345:15)

timers.js:363
      callback.apply(immediate, args);
               ^
TypeError: Cannot call method 'apply' of undefined
    at Object.immediate._onImmediate (timers.js:363:16)
    at processImmediate [as _immediateCallback] (timers.js:345:15)

tedh-shopbot, 2015-04-21:
In the previous ... that was a result of doing a couple of Z moves that worked ... then trying to move X and Y ... then things locked up. After this install, most of the config values were correct except for the power management #'s that were empty ...

tedh-shopbot, 2015-04-21:
And one more log ... got values for X and Y into config, but still no motion ...

info: Checking engine data directory tree... [engine]
info: Loading engine configuration... [engine]
info: Applying engine configuration... [engine]
info: Configuring database... [engine]
debug: Saving config to /opt/fabmo/config/engine.json [config]
info: Databases are clean. [files]
info: Cleaning up database... [engine]
info: Connecting to G2... [engine]
info: Opening control port /dev/ttyACM0 [g2]
info: Dual USB since control port and gcode port are different. (/dev/ttyACM0,/dev/ttyACM1) [g2]
info: Configuring G2... [engine]
debug: Configuring driver... [g2_config]
info: Getting G2 firmware version... [engine]
info: G2 Firmware Build: 83.05 [engine]
info: Loading OpenSBP Commands... [engine]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/Interpolate.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/custom.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/cut.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/experimental.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/file.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/jog.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/leveler.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/move.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/setting.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/value.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/zero.js" [opensbp]
info: Configuring OpenSBP runtime... [engine]
info: Configuring dashboard... [engine]
info: Loading dashboard apps... [engine]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/app_manager" because it already exists. [app_manager]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/configuration" because it already exists. [app_manager]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/editor" because it already exists. [app_manager]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/handwheel" because it already exists. [app_manager]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/job_manager" because it already exists. [app_manager]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/macro_manager" because it already exists. [app_manager]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/network_manager" because it already exists. [app_manager]
debug: Not decompressing app "/opt/fabmo/apps/77b248c0-c67a-11e4-949d-dd104c32afbf.fma" because it already exists. [app_manager]
debug: Not decompressing app "/opt/fabmo/apps/8075f460-c388-11e4-9836-7b1688251980.fma" because it already exists. [app_manager]
debug: Not decompressing app "/opt/fabmo/apps/89416390-c388-11e4-9836-7b1688251980.fma" because it already exists. [app_manager]
debug: Not decompressing app "/opt/fabmo/apps/8d83a3e0-c677-11e4-949d-dd104c32afbf.fma" because it already exists. [app_manager]
info: Loading macros... [engine]
info: Setting up the webserver... [engine]
info: Configuring cross-origin requests... [engine]
info: Cofiguring upload directory... [engine]
info: Loading routes... [engine]
debug:   Loaded routes from "/opt/fabmo/app/routes/config.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/dashboard.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/direct.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/file_commands.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/jobs.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/kill.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/log.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/macros.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/network_manager.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/passthrough.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/websocket.js" [routes]
info: FabMo Engine listening at http://0.0.0.0:9876 [engine]
debug: client connected [websocket]
info: Got a machine state change: manual [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
error: 204,STAT_COMMAND_REJECTED_BY_ALARM,command was not processed because machine is alarmed [machine]
[... 167 lines cut ...]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
[... cut ...]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
war

ryansturmer, 2015-04-21:
Oh!  I think this is actually a known issue that has been resolved.  Shell into your tool and update the engine software, by doing the following after SSHing into the tool.

```
rm -rf /opt/fabmo/config
cd /opt/fabmo/app
git reset --hard HEAD
git pull origin master
```

follow this with a reboot.

Your jerk settings and such will be lost, you'll have to add them back with the config page.

tedh-shopbot, 2015-04-21:
Did not try above yet ... but went back to V300 ... pretty much the same. One move and everything locks up ... will try the above ...

info: Checking engine data directory tree... [engine]
info: Loading engine configuration... [engine]
info: Applying engine configuration... [engine]
info: Configuring database... [engine]
debug: Saving config to /opt/fabmo/config/engine.json [config]
info: Databases are clean. [files]
info: Cleaning up database... [engine]
info: Connecting to G2... [engine]
info: Opening control port /dev/ttyACM0 [g2]
info: Dual USB since control port and gcode port are different. (/dev/ttyACM0,/dev/ttyACM1) [g2]
info: Configuring G2... [engine]
debug: Configuring driver... [g2_config]
info: Getting G2 firmware version... [engine]
info: G2 Firmware Build: 79.6 [engine]
info: Loading OpenSBP Commands... [engine]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/Interpolate.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/custom.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/cut.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/experimental.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/file.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/jog.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/leveler.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/move.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/setting.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/value.js" [opensbp]
debug: Loaded OpenSBP commands from "/opt/fabmo/app/runtime/opensbp/commands/zero.js" [opensbp]
info: Configuring OpenSBP runtime... [engine]
info: Configuring dashboard... [engine]
info: Loading dashboard apps... [engine]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/app_manager" because it already exists. [app_manager]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/configuration" because it already exists. [app_manager]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/editor" because it already exists. [app_manager]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/handwheel" because it already exists. [app_manager]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/job_manager" because it already exists. [app_manager]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/macro_manager" because it already exists. [app_manager]
debug: Not copying app "/opt/fabmo/app/dashboard/apps/network_manager" because it already exists. [app_manager]
debug: Not decompressing app "/opt/fabmo/apps/77b248c0-c67a-11e4-949d-dd104c32afbf.fma" because it already exists. [app_manager]
debug: Not decompressing app "/opt/fabmo/apps/8075f460-c388-11e4-9836-7b1688251980.fma" because it already exists. [app_manager]
debug: Not decompressing app "/opt/fabmo/apps/89416390-c388-11e4-9836-7b1688251980.fma" because it already exists. [app_manager]
debug: Not decompressing app "/opt/fabmo/apps/8d83a3e0-c677-11e4-949d-dd104c32afbf.fma" because it already exists. [app_manager]
info: Loading macros... [engine]
info: Setting up the webserver... [engine]
info: Configuring cross-origin requests... [engine]
info: Cofiguring upload directory... [engine]
info: Loading routes... [engine]
debug:   Loaded routes from "/opt/fabmo/app/routes/config.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/dashboard.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/direct.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/file_commands.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/jobs.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/kill.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/log.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/macros.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/network_manager.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/passthrough.js" [routes]
debug:   Loaded routes from "/opt/fabmo/app/routes/websocket.js" [routes]
info: FabMo Engine listening at http://0.0.0.0:9876 [engine]
debug: client connected [websocket]
debug: Saving config to /opt/fabmo/config/g2.json [config]
info: Got a machine state change: manual [machine]
error: 203,STAT_LIMIT_SWITCH_HIT,a limit switch was hit causing shutdown [machine]
error: 203,STAT_LIMIT_SWITCH_HIT,a limit switch was hit causing shutdown [machine]
error: 203,STAT_LIMIT_SWITCH_HIT,a limit switch was hit causing shutdown [machine]
error: 203,STAT_LIMIT_SWITCH_HIT,a limit switch was hit causing shutdown [machine]
error: 203,STAT_LIMIT_SWITCH_HIT,a limit switch was hit causing shutdown [machine]
error: 203,STAT_LIMIT_SWITCH_HIT,a limit switch was hit causing shutdown [machine]
error: 203,STAT_LIMIT_SWITCH_HIT,a limit switch was hit causing shutdown [machine]
error: 203,STAT_LIMIT_SWITCH_HIT,a limit switch was hit causing shutdown [machine]
[... 201 lines cut ...]
error: 203,STAT_LIMIT_SWITCH_HIT,a limit switch was hit causing shutdown [machine]
warn: WARNING QUIT PENDING WHILE DOING A FIXED MOVE [g2]
error: 203,STAT_LIMIT_SWITCH_HIT,a limit switch was hit causing shutdown [machine]
[... cut ...]
error: 203,STAT_LIMIT_SWITCH_HIT,a limit switch was hit causing shutdown [machine]
error: 203,STAT_LIMIT_SWITCH_HIT,a limit switch was hit causing shutdown [machine]
error: 203,STAT_LIMIT_SWITCH_HIT,a limit switch was hit causing shutdown [machine]
error: 203,STAT_LIMIT_SWITCH_HIT,a limit switch was hit causing shutdown [machine]
error: 203,STAT_LIMIT_SWITCH_HIT,a limit switch was hit causing shutdown [machine]
error: 203,STAT_LIMIT_SWITCH_HIT,a limit switch was hit causing shutdown [machine]
error: 203,STAT_LIMIT_SWITCH_HIT,a limit switch was hit causing shutdown [machine]

tedh-shopbot, 2015-04-21:
Same thing after doing that update ... I dont get the limit switch error ... same thing with and without the limits even plugged in ... one move and it crashes.

ryansturmer, 2015-04-21:
I don't think it has anything to do with limits.

What happens if, on a clean boot, you just use the editor to send it a G-Code, rather than trying to use OpenSBP or the handwheel?  Can you move the tool that way?

gordonab, 2015-04-22:
This may be unrelated: See if you can get the editor to make more that 2 moves. Mine is locking up after 2 moves or jogs.

ryansturmer, 2015-04-22:
My editor (latest master) will do any number of g-codes, but won't do any OpenSBP commands, owing to some sort of error with the transforms stuff in the runtime (missing ptData in the transforms settings)

ryansturmer, 2015-09-29:
This is long since resolved, units play much nicer now.

closed 2015-09-29 by ryansturmer