FabMo/FabMo-Engine #101: OpenSBP VS command sets max feedrate

ryansturmer, 2015-04-21:
Currently, the VS ([V]alue - [S]peed settings) command sets the feedrate limit on G2.  This is probably not what we want.  VS should effectively control the feedrate for moves, but shouldn't actually set the hard upper limit in G2.  Setting the JOG speed should set the G2 jog limit, but move speeds are drawn from an internal OpenSBP variable and probably shouldn't have to change G2 settings at all.

gordonab, 2015-04-21:
So, how do we want to set upper limits for moves? What if the speed setting in VS is higher than the limit for an axis? Would it be better to set the speed limits on your config page? I can make the VS command set the current move speed for use in sending gcode commands only and not have any effect on the g2 settings at all. Another option is to expand the VS parameter list to include the speed limits after the current speed settings.

ryansturmer, 2015-04-21:
The speed limits are already settable in the config page, and I think that's sufficient for now.  If the speed limit is 5ips and you try to move 6, you just get 5, so yes, in my view, just using VS to set the speed for moves in the emitted gcodes is sufficient. @tedh-shopbot might want to weigh in here.

 We might want to expand the command for the future - but I'd like to avoid "upgrading" the OpenSBP language itself wherever possible for now, and instead focus on getting the current functionality working the way we expect.  It will be much easier to test if we can compare it one-to-one with what we currently run in sb3.  Is that a sensible strategy?

gordonab, 2015-04-21:
That makes sense. We'll have to document this well in order to avoid lots
of tech support on "my tool won't go the speed I sent it".  I can have this
fixed shortly.

ryansturmer, 2015-04-21:
Are there any move speed limits in the current shopbot software?  Seems like there must be a maximum somewhere - is it adjustable?

gordonab, 2015-04-21:
The move speed limits are set in the PRM file and aren't adjustable by the user. We should still have some way of setting practical limits. We do this now primarily for safety sake. This is especially true for the handwheel. We currently use the guidelines for robotic teach mode to determine the limits of the keypad since this it the most likely time the the user may be physically in the direct vicinity of the cutting head and cutter during job setup, zeroing and the such. Is there a way to set the limits still in the PRM like JSON file and make sure that they don't go over those settings, but have the flexibility of setting the limits on their tool. This may be handy for setting up a Gantry tool differently than a HandiBot.

gordonab, 2015-04-21:
If I knew what those limits were when in the VS command I could throw an error for being over the limit.

tedh-shopbot, 2015-04-21:
At the risk of adding confusion ...
I think we want to distinguish between "user settings" and "developer configuration settings". To be practical and consistent. "User settings" are the ones we are currently able to set in Sb3 (we can discuss at some later point whether we want to modify of re-categorize these); "developer configuration settings" are the ones that we consider beyond the typical user and only adjustable through the interface or directly in G2. These would include of things like the speed limits set in ShopBot's PRM file, as well as new things like the driver current settings (very scary to let users set, but they are there). For the move or jog speeds, in the current Sb3, if you try and set them beyond the limit, you get a warning and they are set to the limit (if it does not crash it :( ...)

ryansturmer, 2015-04-28:
An interface I've seen for dealing with BAD DANGEROUS settings is to have a settings interface that you have to "unlock" - on the mac, this is the cute euphemism they wrap around the notion of root access.  When you're in a control panel, and you need to do something "dangerous" or something that requires elevated security clearance, you get something like this:

[image]
(Bottom right corner)

I'm always in favor "trust the user" but I do say that in quotes.  A more detailed opinion might be.  "Trust the user: but make sure they know damn well the consequences of their actions."  The little padlock icon mechanic is (I think) a clever way to do that without the user feeling encumbered.  (Certainly there are going to be plenty of instances where a single user will have a perfectly legitimate reason to change something that _most_ users shouldn't mess with.

tedh-shopbot, 2015-04-28:
Well ... that's ok. But for the moment, I see no problem just leaving the relatively hard coded items in tables only available by going into the system (or made available to us as per some of the recent configuration pages). Otherwise I like having them accessible through programming or the standard configuration interface. I'm mostly just interested in having a reasonable subset that is programmable in the OpenSBP language or some such thing ...

ryansturmer, 2015-04-28:
Many are already programmable via the OpenSBP language - if you can list the higher priority ones, we can get them incorporated quickly (they are very easy to add command handlers/system vars for)

ryansturmer, 2016-02-02:
@gordonab Is this issue resolved in the OpenSBP language?  I feel like this was resolved by our decisions about configuration in OpenSBP that we made a month or so ago.

gordonab, 2016-02-29:
This only sets the move speeds in opensbp.json and does not change the max in g2.json.

closed 2016-02-29 by gordonab