FabMo/FabMo-Engine #103: Dashboard Improvement Details ...

tedh-shopbot, 2015-04-27:
For when we start working on refining:
(Maybe we should create a formal wish list for Dashboard to share with Clarke, etc ...)
    - improve side panel behaviors, esp in mobile
    - reconcile "back" function behavior w/in Dashboard
    - display location in top bar in all size views (make sure it expands well to 6 axes and handle large numbers)
    - titles of app manager and job manager ... probably should not both be "manager" (maybe app configuration)
    - progress bar not working in job manager
    - need to close hand-wheel when running a file and displaying DRO (replace with something interesting for file running?)

ryansturmer, 2015-04-28:
1. __Side Panel Behavior:__ What are the behaviors we want here?  Anything that is stand-out terrible? (both on mobile and on the desktop)
2. __Back Function:__ I have done *some* work on this already - it works better than it did originally - what again should that be?
3. __App/Job Manager:__  Suggested titles?  I think they're appropriate, but it could be "My Apps" and "My Jobs" (I always felt those sound cheesy)  - I want to distinguish "Apps" from "App Manager" or "App Configuration" - It's interesting though, because the App manager really isn't about configuring the apps, it's about configuring which apps are installed/enabled (which is why I call it a manager, which I guess isn't really a better term)
4. __Progress Bar__ I'm assuming you mean in the OpenSBP environment.  I don't actually like the progress bar as a feature with OpenSBP, because it's guaranteed not to be accurate in instances that contain looping etc.  What do we see as its future?
5. __Handwheel disable while run:__ No comment here, will open an issue for this.

tedh-shopbot, 2015-04-28:
1 & 2 ... a little easier to describe by showing. 2 yes, better
3. Problem is with using the word "manager" for both of them ... it make it too confusing as both are sort of ambiguous (and especially since in some cases the Job one is called Job Queue and others Job Manager). How about lets use Job Manager and App _____(<> manager)_____.
4. I meant the one at the top of the Job Manager (it previously worked?, now it does not). Actually, I think it works out that just reporting % in terms of lines of file executed works pretty well ... and it's about the only quick way to provide an estimate of what's left in a file. I think people understand that it will not take loops into account. But if that worries ... we could do a more refined count.

ryansturmer, 2016-02-02:
Closing this, as most of these are out of date.  We can reopen *single* issues as needed.

closed 2016-02-02 by ryansturmer