FabMo/FabMo-Engine #104: Handling SBP modal ...

tedh-shopbot, 2015-04-27:
Got the following error on first move from 0,0 after start-up ... did not check to see if it happens after later moves.

M2, 0, 3        ... works fine
M2,  , 3         ... fails; as far as I could tell, both engine and v300 had to be rebooted

gordonab, 2015-04-28:
I found the error in the M2 command when and modal command is sent. This is fixed, though the parser isn't removing any spaces that are sent in the command. If there are just commas without anything in between the command will work. But, if there are spaces the parser is sending a space as the parameter. This space is then being sent to emit_move that is trying to toFixed the space and failing. Can the parser be modified to remove the spaces and send the parameters as undefined? I'll have all of the rest of the move and jog commands handling locations that aren't sent modally.

gordonab, 2015-04-28:
I have the move/jog commands handling spaces in the args as well. This may not be the most efficient way of handling modal parameters, but it is working.

tedh-shopbot, 2015-04-28:
Not sure I quite understand the above. But is seems to me that in OpenSBP all spaces and white space should be ignored (including any non-printing characters). *that is with the exception of possibly supporting the original space/comma equivalence for the first position after the two command letters.

gordonab, 2015-04-28:
I've gotten most of the possible cases worked out, but this one:

M3,1 , , 

is giving me a hard time! Or any move that has a comma as the last character.

gordonab, 2015-04-28:
error: Syntax Error on line 0 [sbp]
error: Expected [object Object],[object Object] but found , [sbp]
Here's the error:

error: 1 [sbp]
UNCAUGHT EXCEPTION
undefined

C:\Users\gordon\Documents\GitHub\FabMo-Engine\runtime\opensbp\opensbp.js:79
                throw err
                      ^
SyntaxError: Expected [ \t] or end of input but "," found.

ryansturmer, 2015-04-28:
Parser now collapses arguments that are all whitespace to an 'undefined' value.  This should make it easier on the command handlers as they won't have to catch all those whitespace edge cases:

Change is here:
https://github.com/ShopBotTools/FabMo-Engine/commit/21b87229102dc6ec0047e693aec78358dae52800

gordonab, 2015-04-28:
> 21b8722

It isn't returning what you are expecting all of the time. I am getting
several different types of parameters in the move functions. They could be
any of the following depending on the spaces and missing parameters.

null
""
" "
"  "

undefined occasionally passed as well.

I'll work up better examples from the log in the morning.

ryansturmer, 2015-04-28:
Updated again, to improve whitespace insensitivity. Pull again before producing your example, I think the current code should have it mostly worked out.

ryansturmer, 2015-04-28:
@tedh-shopbot Can you explain here exactly how that initial space works, with some examples?  I'm sure I don't handle it correctly, as I don't fully understand the intent.

gordonab, 2015-04-28:
The space after a command is used the same way as a comma after the command. For example: M3 1,2,3 is the same as M3,1,2,3. But, M3 ,1,2,3 is an invalid command. This last example would be the same as and M4 command. Another example: VS,1,1 === VS 1,1 or M4,1,2,3,4 === M4 1, 2, 3, 4

If the comma is right after the second command character, any spaces that follow are disregarded and commas rule the format of the parameters. This making the only possible space of any consequences, if it is there, the one after the command. This applies to any OpenSBP command that can be input from the Editor or in a file.

This is a legacy issue more than one we should advertise going forward. Though, it's this legacy concern that means we should still maintain the convention. How hard is to the cull the commandline and look for this space?

gordonab, 2015-04-28:
Other than the first space issue, the rest of the whitespace issues seem to have been handled.

ryansturmer, 2015-10-16:
Resolved in d07f4dd

closed 2015-10-16 by ryansturmer