FabMo/FabMo-Engine #105: Not sure about this one ... but is seems like ...

tedh-shopbot, 2015-04-27:
In a a toolpath that involves pocketing a circle with a short step-over between each pocketing pass as in this view:
[image]

[This one is about 2 inches, done with a Handibot @ 1ips] It seems that at each step-over there is no ramping. The tool just bangs into and out of the step-over. Adjusting the jerk from 2-24 had no effect on the motion. Before we bother Alden, am I missing some fine-tuning that needs to be done in the set-up?

ryansturmer, 2015-04-28:
Ted, 

Thanks for the screenshot!  Can you attach both the OpenSBP file, and if you can post the same toolpath to G-Code and see if you get the same effect, that would be helpful too.  There is a possibility that this is an OpenSBP-specific issue, but I'm not sure what that would look like unless there was a stack break ocurring in between the steps.  Is there a speed change for step-overs?

tedh-shopbot, 2015-04-28:
'Here's SBP FILE
'==========================================
'SHOPBOT FILE IN INCHES
'Turning router ON
SO,1,1
PAUSE 2
'
'Toolpath Name = OutsidePocket
'Tool Name   = 1/8" Straight (13728)
MS,1.0,0.5
JZ,0.800000
J2,0.000000,0.000000
J3,0.000000,0.005000,0.200000
M3,0.000000,0.005000,-0.080000
M3,-0.000725,0.004947,-0.080000
M3,-0.001469,0.004779,-0.080000
M3,-0.002206,0.004487,-0.080000
M3,-0.002906,0.004069,-0.080000
M3,-0.003536,0.003536,-0.080000
M3,-0.004069,0.002906,-0.080000
M3,-0.004487,0.002206,-0.080000
M3,-0.004779,0.001469,-0.080000
M3,-0.004947,0.000725,-0.080000
M3,-0.005000,0.000000,-0.080000
M3,-0.004947,-0.000725,-0.080000
M3,-0.004779,-0.001469,-0.080000
M3,-0.004487,-0.002206,-0.080000
M3,-0.004069,-0.002906,-0.080000
M3,-0.003536,-0.003536,-0.080000
M3,-0.002906,-0.004069,-0.080000
M3,-0.002206,-0.004487,-0.080000
M3,-0.001469,-0.004779,-0.080000
M3,-0.000725,-0.004947,-0.080000
M3,0.000000,-0.005000,-0.080000
M3,0.000658,-0.004957,-0.080000
M3,0.001333,-0.004819,-0.080000
M3,0.002008,-0.004579,-0.080000
M3,0.002658,-0.004235,-0.080000
M3,0.003260,-0.003791,-0.080000
M3,0.003791,-0.003260,-0.080000
M3,0.004235,-0.002658,-0.080000
M3,0.004579,-0.002008,-0.080000
M3,0.004819,-0.001333,-0.080000
M3,0.004957,-0.000658,-0.080000
M3,0.005000,0.000000,-0.080000
M3,0.004957,0.000658,-0.080000
M3,0.004819,0.001333,-0.080000
M3,0.004579,0.002008,-0.080000
M3,0.004235,0.002658,-0.080000
M3,0.003791,0.003260,-0.080000
M3,0.003260,0.003791,-0.080000
M3,0.002658,0.004235,-0.080000
M3,0.002008,0.004579,-0.080000
M3,0.001333,0.004819,-0.080000
M3,0.000658,0.004957,-0.080000
M3,0.000000,0.005000,-0.080000
M3,0.000000,0.067500,-0.080000
M3,-0.009782,0.066787,-0.080000
M3,-0.019832,0.064521,-0.080000
M3,-0.029786,0.060573,-0.080000
M3,-0.039225,0.054933,-0.080000
M3,-0.047730,0.047730,-0.080000
M3,-0.054933,0.039225,-0.080000
M3,-0.060573,0.029786,-0.080000
M3,-0.064521,0.019832,-0.080000
M3,-0.066787,0.009782,-0.080000
M3,-0.067500,0.000000,-0.080000
M3,-0.066787,-0.009782,-0.080000
M3,-0.064521,-0.019832,-0.080000
M3,-0.060573,-0.029786,-0.080000
[... 965 lines cut ...]
M3,-0.722109,-0.417446,-0.175000
M3,-0.722107,-0.417480,-0.175000
M3,-0.722109,-0.417515,-0.175000
M3,-0.722139,-0.417662,-0.175000
M3,-0.722192,-0.417810,-0.175000
M3,-0.722268,-0.417955,-0.175000
M3,-0.722368,-0.418090,-0.175000
M3,-0.722488,-0.418209,-0.175000
M3,-0.722623,-0.418309,-0.175000
M3,-0.722767,-0.418386,-0.175000
M3,-0.722916,-0.418438,-0.175000
M3,-0.723062,-0.418468,-0.175000
M3,-0.723097,-0.418471,-0.175000
M3,-0.723132,-0.418468,-0.175000
M3,-0.723278,-0.418438,-0.175000
M3,-0.723427,-0.418386,-0.175000
M3,-0.723572,-0.418309,-0.175000
M3,-0.723707,-0.418209,-0.175000
M3,-0.723826,-0.418090,-0.175000
M3,-0.723926,-0.417955,-0.175000
M3,-0.724002,-0.417810,-0.175000
M3,-0.724055,-0.417662,-0.175000
M3,-0.724067,-0.417603,-0.175000

END

tedh-shopbot, 2015-04-28:
(AND G CODE)
(===============================)

G17
G0Z0.8000
G0X0.0000Y0.0000S14000
M3
M8
(J3)
G0X0.0000Y0.0050Z0.2000F120.0
G1Z-0.0800F30.0
G1X-0.0007Y0.0049F60.0
G1X-0.0015Y0.0048F60.0
G1X-0.0022Y0.0045F60.0
G1X-0.0029Y0.0041F60.0
G1X-0.0035Y0.0035F60.0
G1X-0.0041Y0.0029F60.0
G1X-0.0045Y0.0022F60.0
G1X-0.0048Y0.0015F60.0
G1X-0.0049Y0.0007F60.0
G1X-0.0050Y0.0000F60.0
G1X-0.0049Y-0.0007F60.0
G1X-0.0048Y-0.0015F60.0
G1X-0.0045Y-0.0022F60.0
G1X-0.0041Y-0.0029F60.0
G1X-0.0035Y-0.0035F60.0
G1X-0.0029Y-0.0041F60.0
G1X-0.0022Y-0.0045F60.0
G1X-0.0015Y-0.0048F60.0
G1X-0.0007Y-0.0049F60.0
G1X0.0000Y-0.0050F60.0
G1X0.0007Y-0.0050F60.0
G1X0.0013Y-0.0048F60.0
G1X0.0020Y-0.0046F60.0
G1X0.0027Y-0.0042F60.0
G1X0.0033Y-0.0038F60.0
G1X0.0038Y-0.0033F60.0
G1X0.0042Y-0.0027F60.0
G1X0.0046Y-0.0020F60.0
G1X0.0048Y-0.0013F60.0
G1X0.0050Y-0.0007F60.0
G1X0.0050Y0.0000F60.0
G1X0.0050Y0.0007F60.0
G1X0.0048Y0.0013F60.0
G1X0.0046Y0.0020F60.0
G1X0.0042Y0.0027F60.0
G1X0.0038Y0.0033F60.0
G1X0.0033Y0.0038F60.0
G1X0.0027Y0.0042F60.0
G1X0.0020Y0.0046F60.0
G1X0.0013Y0.0048F60.0
G1X0.0007Y0.0050F60.0
G1X0.0000Y0.0050F60.0
G1Y0.0675F60.0
G1X-0.0098Y0.0668F60.0
G1X-0.0198Y0.0645F60.0
G1X-0.0298Y0.0606F60.0
G1X-0.0392Y0.0549F60.0
G1X-0.0477Y0.0477F60.0
G1X-0.0549Y0.0392F60.0
G1X-0.0606Y0.0298F60.0
G1X-0.0645Y0.0198F60.0
G1X-0.0668Y0.0098F60.0
G1X-0.0675Y0.0000F60.0
G1X-0.0668Y-0.0098F60.0
G1X-0.0645Y-0.0198F60.0
G1X-0.0606Y-0.0298F60.0
G1X-0.0549Y-0.0392F60.0
G1X-0.0477Y-0.0477F60.0
G1X-0.0392Y-0.0549F60.0
[... 1745 lines cut ...]
G1X-0.1241Y-0.0182F60.0
G1X-0.1199Y-0.0369F60.0
G1X-0.1126Y-0.0554F60.0
G1X-0.1021Y-0.0729F60.0
G1X-0.0887Y-0.0887F60.0
G1X-0.0729Y-0.1021F60.0
G1X-0.0554Y-0.1126F60.0
G1X-0.0369Y-0.1199F60.0
G1X-0.0182Y-0.1241F60.0
G1X0.0000Y-0.1254F60.0
G1X0.0182Y-0.1241F60.0
G1X0.0369Y-0.1199F60.0
G1X0.0554Y-0.1126F60.0
G1X0.0729Y-0.1021F60.0
G1X0.0887Y-0.0887F60.0
G1X0.1021Y-0.0729F60.0
G1X0.1126Y-0.0554F60.0
G1X0.1199Y-0.0369F60.0
G1X0.1241Y-0.0182F60.0
G1X0.1254Y0.0000F60.0
G1X0.1243Y0.0165F60.0
G1X0.1209Y0.0335F60.0
G1X0.1149Y0.0504F60.0
G1X0.1062Y0.0667F60.0
G1X0.0951Y0.0818F60.0

ryansturmer, 2015-04-28:
@aldenhart Any thoughts on this?  I think this may be a genuine G2 issue (or a misconfiguration on our part)

ryansturmer, 2015-05-01:
Here's a video of the behavior in action:

https://youtu.be/Q_tVzYJN7ck

ryansturmer, 2016-01-08:
This turned out to be ja/jd related, if I recall.

closed 2016-01-08 by ryansturmer