FabMo/FabMo-Engine #113: Dimmed area in main menu is weird and should feel bad about itself.

ryansturmer, 2015-05-01:
When the right side DRO menu is displayed, the main UI is "dimmed" such that any click in the main UI dismisses the DRO.  When the main menu is visible, the dimming doesn't go all the way to the bottom of the page, which looks weird, and doesn't quite work the way we intend:

[image]

Screenshot thanks to @tedh-shopbot

wholetthedogsout, 2015-09-07:
This looks closed

ryansturmer, 2015-09-08:
Still happens on my android device.

wholetthedogsout, 2015-09-08:
Sorry this is fixed on the branch I made a pull request for. Would you like to me to merge them?

ryansturmer, 2015-09-08:
Yes by all means!  Don't break the build ;)

closed 2015-09-08 by wholetthedogsout