FabMo/FabMo-Engine #115: FabMo V9 for maker faire

tedh-shopbot, 2015-05-09:
First of the MF FabMo Handibots is more or less working. I think I am having trouble with that spindle output because I was just expecting to be able to turn it on with a command rather than inserting a command in a file. The g-code version blinks and instantly shuts off if you just send it. The ShopBot version does that the first time, then it will turn on and stay on ... but crashes the software.

I'll be back tomorrow afternoon to do some more puttering. Not for MF, but in the long run are we planning to use a permissive switch as with SB3?

tedh-shopbot, 2015-05-10:
OK ... here's more detail. Output switches work in a file. But sending a switch ON command only blinks the output in gcode and blinks the first time in ShopBot and then crashes G2 the second time.

ryansturmer, 2015-10-13:
Newer G2 revisions fix that.

closed 2015-10-13 by ryansturmer