FabMo/FabMo-Engine #119: Changing the way wifi connection work.

jlucidar, 2015-06-16:
As suggested by ryan in our last discussion, We have to change the way wifi connections are handle.

There is a pretty neat tool out there called connman that should be easy to integrate with node.js.  
There is a npm module [there](https://www.npmjs.com/package/connman-simplified).

I suggest that we adopt this pattern ;
- the machine tries to connect to a known network
- if it doesn't succeed, it turns into a wifi hotspot.
- in hotspot mode, you can add a network through the dashboard.

I'm working on it tonight.

jlucidar, 2015-07-22:
here is an illustration of something I can do with the module : 
[image]

@tedh-shopbot @gordonab @ryansturmer any suggestion on that point ?

MatthewHux, 2015-07-22:
Would there be a provision for changing the wireless network? For instance you may want to go over to your neighbors house to work on a project. It automatically connects to your wireless network since it is still in range, but it is weak. You will then want to change it to log into your neighbors.

Also provisions for when it finds two known networks. Either have a user set priority or have it connect to the strongest signal?

jlucidar, 2015-07-24:
> Would there be a provision for changing the wireless network? For instance you may want to go over to your neighbors house to work on a project. It automatically connects to your wireless network [...]

Well, This is something quite hard to deal with... As it's connected, the machine will not look for another network (more strong) to connect to. this is something feasible but I don't know if that's the best thing to do. If you're exactly between two networks, your machine will switch in loop from one to the other...
An other option may to add a button that allow you to turn on the hotspot and you can choose your network from there...

>Also provisions for when it finds two known networks. Either have a user set priority or have it connect to the strongest signal?

I guess it will choose the strongest signal. it takes the first network that it list actually.

closed 2016-01-21 by ryansturmer