FabMo/FabMo-Engine #131: Apps not installed in approot when starting with systemctl enable and start

gordonab, 2015-07-30:


ryansturmer, 2015-08-15:
Is this still an issue?  Can you post steps to reproduce if so?

tedh-shopbot, 2015-08-15:
I think if we make up some more systems, we'll get back to this issue ... probably can't recreate exactly until Gordon comes back. We did not spend much time on it because the edison version we are running is not set up with the same protective? image as the BBB ... ?

ryansturmer, 2015-08-17:
The fact that the edison isn't running the RO root image means that this really shouldn't be an issue at all!  So if we're seeing it, there's likely a problem.

-R

ryansturmer, 2015-09-12:
Anyone seen this lately?  @tedh-shopbot @gordonab

tedh-shopbot, 2015-09-12:
nope

gordonab, 2015-09-12:
That last couple units I've tried this on worked.

- Gordon

closed 2015-10-13 by ryansturmer