FabMo/FabMo-Engine #139: Lag in updating position on phone

wlyoung, 2015-08-11:
Hey guys,

We ran a FabMo Handibot at Fab11 and could connect fine, but there was some lag in updating the position in both the location and the line-by-line display of the file...sometimes a second or two. It seemed to happen more when the file restarted the loop, but also happened in the middle of the file on occasion. 

I don't remember it happening as much when we were connected through the wireless router as when we were connected directly via wifi?

jlucidar, 2015-08-11:
It might be a problem with the Wifi signal strength . It probably depends on the distance between the router and the machine.
I had several time the same issue when I was controlling the machine through the router (the machine was at about 20ft from it). Moving the machine a couple of inches closer to the router could change drastically the situation.

Are you using the intel edison with the external wifi antenna ?

ryansturmer, 2015-08-14:
I would not write this off to wifi reception - I think this may be an issue of a dropped websocket.  I have seen the same thing in situations where i was using hardwired ethernet.

closed 2016-01-08 by ryansturmer