FabMo/FabMo-Engine #146: network manager should use POST for endpoints that modify the system state

ryansturmer, 2015-08-25:
Things like enabling/disabling AP mode, switching networks, disconnecting, forgetting, etc all modify the state of the server, and should be implemented as POST endpoints rather than GET endpoints.  GET can be used for things like fetching lists of networks, indicating device capabilities, etc.

        server.get('/network/wifi/on',turnWifiOn);//OK
        server.get('/network/wifi/off',turnWifiOff);//Ok
        server.get('/network/hotspot/on',turnWifiHotspotOn);//OK
        server.get('/network/hotspot/off',turnWifiHotspotOff);//Ok        
        server.get('/network/wifi/disconnect',disconnectWifi); //OK

The above endpoints (at least) should all be POST.

jlucidar, 2015-08-26:
Would this be more suitable ? 
 ```javascript
server.post('/network/wifi',wifiActions);
server.post('/network/hotspot',hotspotActions);
```
the json you send it would be something like : 
````json
{"enabled":true}
{"enabled":false}
{"disconnect":true}
```

jlucidar, 2015-08-26:
Or something more like : 
```javascript
server.post('/network/wifi/state',wifiState);
server.post('/network/wifi/disconnect',disconnectWifi);
server.post('/network/hotspot'/state,hotspotState);
``` 
?

ryansturmer, 2015-08-26:
I think having the individual endpoints for connect/disconnect/forget is nice.  It's more self-documenting, and it's consistent with what we've done elsewhere in the API

closed 2015-08-26 by jlucidar