FabMo/FabMo-Engine #149: Units in DRO

ryansturmer, 2015-08-27:
Units should be displayed correctly in the DRO

gordonab, 2015-09-03:
Does the DRO get the unit type from the engine or G2? This doesn't change with a G20/G21 or VD command.

aldenhart, 2015-09-03:
The DRO and units are the reason there are the MPOx parameters available in the status reports. If you monitor these values you can switch back and forth between inches and mm easily.

mpox
mpoy
mpoz
(a,b,c if you need it)

ofsx
ofsy
ofsz

unit

MPOs are ALWAYS in mm and in machine coordinates, regardless of the unit setting. Use the offset setting to factor in the user coordinate space, then the unit setting to convert to the proper units

This way the MPOs can always the used for plotting / graphics regardless of what offsets and units are in play

Alden

ryansturmer, 2015-09-05:
this is resolved.  The units are now conveyed to the dashboard, and the display is updated.

closed 2015-09-05 by ryansturmer