FabMo/FabMo-Engine #155: Add inch/metric button

jw4rd, 2015-09-03:
and/or add support for g20/g21 in SBP files

ryansturmer, 2015-09-03:
Thank you for using the issue tracker.

Inch/metric button is a good idea.  You can do G20/G21 through the editor, but there's an equivalent sbp command - @gordonab can you tell us what it is?

gordonab, 2015-09-03:
VD,,<units:0=inches,1=mm>

This OpenSBP command changes the units with G20 or G21, but does nothing
with any of the displayed info in the Dashboard or previewer.

ryansturmer, 2015-09-04:
both VD and G20/G21 will now work to switch units.  There is a "system default units" configuration option in the config page as of 8287110 that lets you set what the engine defaults to.  You can change units at any time with the above mentioned commands, but to make it stick, you have to set it in the configuration settings.

closed 2015-09-04 by ryansturmer