FabMo/FabMo-Engine #156: Cleaning up unused dependencies.

wholetthedogsout, 2015-09-03:
Remove no longer used dependencies.

wholetthedogsout, 2015-09-03:
@ryansturmer getting error with npm install with dbus@0.2.12  only place i find it in the files is in the npm debug log. Any ideas?

ryansturmer, 2015-09-03:
I would focus on the dashboard dependencies rather than those in the engine... the dbus dependency though is for the wireless connection manager for the intel edison, it probably will crash if you try to install  it.  If you are having difficulty getting the engine to run, you can take connman-simplified out of the package.json temporarily

wholetthedogsout, 2015-09-03:
@ryansturmer Thanks. I just wasn't sure which module in the package.json the dbus was affiliated with.

ryansturmer, 2015-10-13:
I think we've done enough dependency cleanup for now, before having really gutted the thing for unneeded framework stuff and components.

closed 2015-10-13 by ryansturmer