FabMo/FabMo-Engine #159: SBP code does not run from editor

jw4rd, 2015-09-03:


gordonab, 2015-09-03:
What kind of commands are you attempting to execute? Can you send some examples for me to investigate?

jw4rd, 2015-09-03:
these are the commands I tried to execute from the editor
VD,,1
M2,1,1

ryansturmer, 2015-09-04:
Do you have the editor set to OpenSBP mode?

gordonab, 2015-09-04:
At the end of the day, I had trouble as well with VD and setting to mm. It
appears to be locking something up. I'll troubleshoot it further first
thing in the morning.

ryansturmer, 2015-09-04:
Were you having trouble with both VD and G20/21? Or just VD?

The switch to millimeters mode does work (in g-code at least) but the units on the display do not update accordingly.  We can fix this pretty easily.

jw4rd, 2015-09-04:
I did not have a problem executing g20/21

ryansturmer, 2015-09-04:
Fixed in d6a5732736b8e2d764ab9c33e5091883b9f70286 - this was an issue with the way the OpenSBP runtime interpreted state transitions from G2, specifically how it handles G-Code programs that contain no motion commands (For instance a program containing only a single G20/G21)

closed 2015-09-04 by ryansturmer