FabMo/FabMo-Engine #164: FabMo in direct access mode job run time in job manager history = current time - 35min

jw4rd, 2015-09-04:


ryansturmer, 2015-09-05:
When you say job run time, do you mean the time at which the job was run?  Or do you mean the job duration?

wlyoung, 2015-09-05:
When it was run. When a job is run and the history is immediately checked,
it says it was run 35 minutes ago.

ryansturmer, 2015-09-05:
When you say direct access mode, you mean when the tool is acting like an access point and you are connecting to it as a station, correct?

jw4rd, 2015-09-05:
Yes, when connected to the tool, not through a router

ryansturmer, 2015-09-06:
This makes sense.  The intel Edison has no RTC, so when it comes up, I'm not sure what it does about resolving the time if it has no external internet connection.  Frankly, it's surprising that the time is only off by 35 minutes given that.... it's probable that the linux image we're using runs fake-hwclock or a similar tool that caches the local time at shutdown.

So the real issue is "Tool has no idea what time it actually is."

There are 2 fixes for this, both of which I feel like we should implement:

1) The RTC_BAT pin on the edison should be pulled out and connected to a battery (or even better a supercap) to provide reserve power for the edison's RTC.
2) There should be a means by which clients can submit time corrections to the fabmo server.  The server should be able to absorb these corrections and adjust its internal time accordingly.  It's almost a certainty that clients have an internal clock time that can be treated as authoritative, and since the edison (or whatever SBC) is likely not to have that when running exclusively in access point mode, there should be no problem for fabmo to say "Oh, i'm in AP mode, and I've always been in AP mode since boot, so I have no authoritative time reference, so I can treat a client as that authoritative reference and adjust my clock time"

I think the http endpoint can be `/time` which could be retrieved with a `GET` or updated with a `POST` - the fabmo server can make an internal assessment of whether or not a `POST`ed time is to be accepted, and return the appropriate HTTP return code.  The dashboard can include a function that when first loaded, it relays an updated time once.

closed 2016-04-22 by ryansturmer