FabMo/FabMo-Engine #170: With 6 characters axis will wrap to next row in DRO table.

wholetthedogsout, 2015-09-09:
Not sure if this is a common scenario. 
[image]

wholetthedogsout, 2015-09-09:
Sorry 7 characters.

ryansturmer, 2015-09-29:
This still happens, but now the number of decimal places when running in millimeters mode is reduced, making it unlikely.

gordonab, 2015-09-30:
This could still be a problem in Inch mode with a large machine. It isn't that uncommon to have a 3 figure number with 3 decimal places and a sign. For instance, the longest machines we make are 288" long.

gordonab, 2015-09-30:
I would expect 8 characters, including the sign and decimal point with be enough for most coordinates. The exception will be with rotary axes, they can go into 5 or 6 digits. It would make sense to limit the degree display to 2 decimal places and 6 digits for a total of 10 characters.

jlucidar, 2015-10-01:
We probably also need at least 6 digit for the metric system ( something
like xxxx.xx mm is totally possible on big machines... )

closed 2016-04-22 by wholetthedogsout