FabMo/FabMo-Engine #175: gcode 'm8' not working in the editor

jw4rd, 2015-09-10:
and spindle turns off without an 'm8'
this program turns on the router for one second and turns off without 'm8':
m4
g4p1

ryansturmer, 2015-09-11:
The spindle is only allowed to run during a machining cycle.  It turns off automatically when your program is over. This is behaving as intended.

jw4rd, 2015-09-11:
What if you need to turn the router off to do a tool change or some other operation?
m4
g4p1
m8
g4p1
m4
g4p1

ryansturmer, 2015-09-11:
I think we're talking about two separate issues here.

Are you saying that M8 doesn't actually turn the spindle off when in a machining cycle?  Or are you saying that the spindle turns off by itself at the end of a cycle?  Or are both occurring?

M8 should not be required to turn off the spindle at the end of a file.  It *should* work however, to turn the spindle off during a file.

ryansturmer, 2015-09-11:
Woops!

It's M5 to stop the spindle, not M8.  Give that a try :walking:

jw4rd, 2015-09-11:
That works.  Thanks.

closed 2015-09-11 by ryansturmer