FabMo/FabMo-Engine #176: Download file in Job Manager History

jw4rd, 2015-09-10:
After clicking 'dowload file' and then clicking the 'home' icon, the downloaded file is still shown in the window.  Add a menu button or back button after downloading a file in the history?

ryansturmer, 2015-09-11:
I'm not sure I follow.  When you click "download file" does it not download it to your downloads folder?

jw4rd, 2015-09-11:
No, it does not download the file.  The gcode is just displayed in a window when I click download.
I'm using firefox and ubuntu.

eliwilliams, 2015-09-28:
This issue has been partially resolved - downloading works for Chrome and Safari currently, but not for Firefox.

wholetthedogsout, 2015-09-28:
I get a part extension when downloading in firefox, not sure why the download is being interrupted, could be security setting?

eliwilliams, 2015-09-28:
This is what's happening to me too. What you'll see is that the part extension actually contains all of the data of the download. If you open the download link in a new tab, the download actually finishes.

closed 2018-02-06 by ryansturmer