FabMo/FabMo-Engine #185: Display issues in webkit based browsers

ryansturmer, 2015-09-23:
More testing needed here.

ryansturmer, 2015-09-29:
Problems when you dismiss the toolbar. Space is not recovered for use by apps etc:

[image]

I think the correct way to fix this is to ditch the left corner hamburger and always display the side toolbar, just optionally in full or collapsed mode (but never totally gone)

ryansturmer, 2015-09-29:
This seems to be the only major lingering issue on this ticket.

wholetthedogsout, 2015-09-29:
So you are fine with the left hamburger being removed all together?

ryansturmer, 2015-09-29:
Yeah, I think, if the left hamburger wasn't there, and collapse just took it from text labels to none, I think we'd be in really good shape.

giseburt, 2015-09-29:
Hi,

Long time listener first time caller.

I found the two hamburgers confusing. You’re using the same icon to mean two things.

Generally, the hamburger (I’m also head them called “three hot dogs”) is a “Menu” button, which is the left. Technically the top-right is a DRO and Thumbwheel button. I would expect to see what I get on the left (a navigation menu) when I click on a hamburger.

I also agree that the hamburger on the left can/should go away if the navigation menu will be persistent. On a limited screen, I don’t think the navigation menu should be persistent.

One option is to have the navigation menu push the contents down, not in from the side. (The hand wheel+DRO can be in from the side as an overlay, still.)

See http://getbootstrap.com/javascript/ <http://getbootstrap.com/javascript/> :

 

The deciding factor of when to go overlay-from-the-side or push-down-from-the-top is really a matter of the content. If there’s too much content, then you should do down-from-top where scrolling is natural, vs. in-from-the-left and then scroll up and down in the overlay.

Slightly off topic (since this problem is partially caused but the location of the DRO): I think the DRO as a universal overlay object might be better served as a bottom left hovering button, like the Googe +-in-a-circle:

     

Mouse-over on mobile would be first-tap. For this purpose we could just use that to toggle the hand-wheel overlay, which contains the sub-actions like “Kill Job” and “Pause.”

Just a dump of my incoherent thoughts.

-Rob

ryansturmer, 2015-09-29:
Thanks for joining the discussion, Rob!  It's good to hear from you.  @tedh-shopbot Any thoughts?

ryansturmer, 2015-09-29:
It's also occurred to me that there are left-handed/right-handed considerations to be made on mobile for interfaces like the handwheel, which are meant for the thumb.

wholetthedogsout, 2015-09-29:
Left hamburger has been removed but definitely open to further steps

tedh-shopbot, 2015-09-29:
My only real thought here is that as a way of preserving some familiarity for old ShopBotters, having the DRO on the left as it is ... is the best way to look familiar. I think it won't be long before people get over even needing to look at the DRO ... just the small numbers changing should suffice...

ryansturmer, 2015-10-02:
Closing this for now, we can consider mobile usability issues in general on another ticket.

closed 2015-10-02 by ryansturmer