FabMo/FabMo-Engine #186: Max jerk setting not remembered ...

tedh-shopbot, 2015-09-24:
Not remembered when set in the configuration app, tool shut down and restarted ...

ryansturmer, 2015-09-24:
All installations?  Or just that one?

tedh-shopbot, 2015-09-24:
Will test ... that is the only one I had in my room ... =ted

gordonab, 2015-09-24:
I had trouble with a unit yesterday, but thought it was one I missed.

ryansturmer, 2015-09-24:
I am having difficulty reproducing - so please try the following (someone)

1) Open configuration page
2) Set log level to "g2"
3) Switch to Axes page and make a change to the max jerk setting (make sure to switch fields to commit the change)
4) Go to the Debug Log (in the browser, through the sidebar)
5) Paste it here

Should shed some light...

ryansturmer, 2015-09-24:
Make sure to set your log level back to "informational" when this is complete.  It shouldn't be left at g2 except for debugging.

gordonab, 2015-09-25:
This is from fabmo_ed_14 on the DT in the Fabmo lab.

info: Checking engine data directory tree... [engine]
info: Loading engine configuration... [engine]
info: Getting engine version... [engine]
debug: Executing shell command: git rev-parse --verify HEAD [util]
info: Engine version: 68d59c9fb3a4fdb72c6223a020cda625e96ffa63 [engine]
info: Applying engine configuration... [engine]
warn: Skipping network setup because wifi manager is disabled. [engine]
info: Configuring database... [engine]
info: Databases are clean. [files]
info: Cleaning up database... [engine]
info: Connecting to G2... [engine]
info: Opening control port /dev/ttyACM0 [g2]
info: Dual USB since control port and gcode port are different.
(/dev/ttyACM0,/dev/ttyACM1) [g2]
info: Loading the machine configuration... [engine]
info: Configuring G2... [engine]
debug: Configuring driver... [config]
error: SyntaxError: Unexpected end of input
    at Object.parse (native)
    at null.<anonymous> (/fabmo/config/config.js:45:16)
    at fs.js:272:14
    at /fabmo/node_modules/fs-extra/node_modules/graceful-fs/graceful-fs.js:42:10
    at Object.oncomplete (fs.js:108:15) [config]
warn: Problem loading the user configuration file
"/opt/fabmo/config/g2.json": Unexpected end of input [config]
debug: finished [config]
info: Getting G2 firmware version... [engine]
info: G2 Firmware Build: 83.09 [engine]
info: Applying machine configuration... [engine]
info: Changing default units to INCH [machine_config]
info: Loading OpenSBP Commands... [engine]
debug: Loaded OpenSBP commands from
"/fabmo/runtime/opensbp/commands/custom.js" [opensbp]
debug: Loaded OpenSBP commands from
"/fabmo/runtime/opensbp/commands/cut.js" [opensbp]
debug: Loaded OpenSBP commands from
"/fabmo/runtime/opensbp/commands/experimental.js" [opensbp]
debug: Loaded OpenSBP commands from
"/fabmo/runtime/opensbp/commands/file.js" [opensbp]
debug: Loaded OpenSBP commands from
"/fabmo/runtime/opensbp/commands/jog.js" [opensbp]
debug: Loaded OpenSBP commands from
"/fabmo/runtime/opensbp/commands/leveler.js" [opensbp]
debug: Loaded OpenSBP commands from
"/fabmo/runtime/opensbp/commands/move.js" [opensbp]
debug: Loaded OpenSBP commands from
"/fabmo/runtime/opensbp/commands/setting.js" [opensbp]
debug: Loaded OpenSBP commands from
"/fabmo/runtime/opensbp/commands/value.js" [opensbp]
debug: Loaded OpenSBP commands from
"/fabmo/runtime/opensbp/commands/zero.js" [opensbp]
info: Configuring OpenSBP runtime... [engine]
error: SyntaxError: Unexpected end of input
    at Object.parse (native)
    at null.<anonymous> (/fabmo/config/config.js:45:16)
    at fs.js:272:14
    at /fabmo/node_modules/fs-extra/node_modules/graceful-fs/graceful-fs.js:42:10
    at Object.oncomplete (fs.js:108:15) [config]
warn: Problem loading the user configuration file
"/opt/fabmo/config/opensbp.json": Unexpected end of input [config]
info: Configuring dashboard... [engine]
info: Loading dashboard apps... [engine]
debug: Not copying app "/fabmo/dashboard/apps/app_manager.fma" because
it already exists. [app_manager]
debug: Not copying app "/fabmo/dashboard/apps/configuration.fma"
because it already exists. [app_manager]
debug: Not copying app "/fabmo/dashboard/apps/editor.fma" because it
already exists. [app_manager]
[... 59 lines cut ...]
info: FabMo Engine listening at http://0.0.0.0:9876 [engine]
debug: client connected [websocket]
debug: client connected [websocket]
debug: Saving config to /opt/fabmo/config/engine.json [config]
g2: --C-1442937385339----> {"xjm":100} [g2]
g2: <-C--1442937385343---- {"r":{"xjm":100},"f":[1,0,12]} [g2]
g2: <-C--1442937385593---- {"sr":{"stat":4}} [g2]
g2: --C-1442937388191----> {"yjm":100} [g2]
g2: <-C--1442937388195---- {"r":{"yjm":100},"f":[1,0,12]} [g2]
g2: <-C--1442937388456---- {"sr":{"stat":4}} [g2]
g2: --C-1442937391966----> {"zjm":100} [g2]
g2: <-C--1442937391970---- {"r":{"zjm":100},"f":[1,0,12]} [g2]
g2: <-C--1442937392220---- {"sr":{"stat":4}} [g2]

Best I can tell from looking at the log and watching the journalctl.
The G2.json file is being erased. The default jerk is 24, the default
speeds are the same as those we are setting in the configuration page,
so it doesn't look like anything is happening to them.

After power cycling the DT, /opt/fabmo/config/g2.json file is empty.
Even after making changes to the jerk setting this file is still empty
as though it was never getting written. Haven't chased that rabbet
yet. But, the log above doesn't indicate that it is.

-Gordon

tedh-shopbot, 2015-09-26:
Problem persists, somewhat differently, on the next one I am trying. This is a bigger problem. After doing a new pull using the standard procedure and restarting, fabmo seems to run fine (wrong jerk, but I can live with that). Then, every time I unplug and restart the configuration file is corrupted. It is corrupted as if the it thinks it is in the wrong units. I have not been able to make this one usable.

ryansturmer, 2015-09-27:
So, what seems to be happening is the files that are being written aren't getting flushed to disk for some reason, and get written as blank (zero-length) files.  When the system boots the next time, it chokes on the malformed files, and doesn't load its default configuration properly.   I am fixing the behavior of choking on the malformed files, but it doesn't explain why the files are getting blanked in the first place, other than just power-cycle disk corruption issues, which seems less likely for as often as it seems to happen.

ryansturmer, 2015-09-27:
Code should recover correctly from damaged config files since 375f98  - This doesn't mean that it will never forget the settings in case of a bad configuration file - there's no way to prevent that.  It does mean that it will recover gracefully from this situation by subbing in the defaults, and should behave much better now.

closed 2015-09-27 by ryansturmer