FabMo/FabMo-Engine #200: Questions about macro routine for zeroing ...

tedh-shopbot, 2015-10-10:
It seems to work well ...
- I'm wondering where the offsets from the switch are stored (the positioning move after hitting). Maybe they should go in the Machine settings section.
- The routine needs to zero once it makes the offsetting moves (or, when you start a file, it just moves back to zero and crashes because it is sitting on the prox switch.
- We will also need the equivalent of C2, the Z-zero setting routine.

ryansturmer, 2015-10-11:
The offsets are stored in the macro itself, for now.

Assigning this to @gordonab - this is an OpenSBP programming exercise.  (And a good chance to squash bugs in the interpreter, etc)

ryansturmer, 2015-10-11:
( @gordonab you're of course free to decline :-) )

ryansturmer, 2015-10-13:
Duplicate of #227

closed 2015-10-13 by ryansturmer