FabMo/FabMo-Engine #206: Right DRO Panel covering or not covering content ...

tedh-shopbot, 2015-10-10:
I think I did not appreciate the earlier discussion of this. It seemed when working today that it made sense for the DRO panel to cover things on a mobile device. There is just not enough room to always have it up.

On a larger screen though, I wanted to leave it in position, but it seemed to be inappropriately covering stuff ...

tedh-shopbot, 2015-10-11:
I thought for a while that this was related to just being full screen on PC ... but seems more idiosyncratic with respect to sequence of openings and closings of apps and right panel ...

ryansturmer, 2015-10-13:
This could be resolved with a "pin" button (this is a suggestion by bill young) where the default behavior is to cover, but the DRO could be "pinned" so that it doesn't ever cover.  You would pin the DRO on desktop more than likely, but leave it be on mobile.  The default behavior could be pinned, unpinned, or selected automatically based on media queries?

wholetthedogsout, 2015-10-20:
The current behavior on the DRO is if you run a file then the DRO will shift the main window contents over, if you then close it with the hamburger it leaves a large gap and doesn't shift back over. If you just open it with the hamburger it will cover the main window contents. Once we figure out the behavior we want I can fix this.

ryansturmer, 2015-10-20:
With the current behavior, the DRO doesn't *always* cover the main content.  It seems like under some media queries it does, and some it doesnt.  I think on a sufficiently small screen (phone, tablet) we want the DRO to cover.  On a larger screen, we want it to shove the content aside so you can see everything.    I'm not sure where the switchover should happen.

tedh-shopbot, 2016-04-22:
this is old enough ... probably best to wait and see if it is still troublesome

closed 2016-04-22 by tedh-shopbot