FabMo/FabMo-Engine #210: Progress bar does not work with a file that is an OpenSBP [CR].

tedh-shopbot, 2015-10-10:


ryansturmer, 2015-10-11:
What is the expected behavior of the progress bar on a file that incorporates flow control, and in files that call other files and/or macros?  Some considerations:

* The progress reflect the current position in the file, basically ignoring any notion of looping.  So the progress would be simply be `CurrentLocation/TotalLines` - This would result (in the case of looping files) in a progress bar that possibly moved back and forth as it progressed, as the line number in the file would not be monotonically increasing
* Progress could go unreported for files that contained *any* flow control constructs.  We could simply assume any file with a loop is a "utility" file and address things that way.  The progress bar could change to a "busy indicator" in those instances.
* For files that nest (such as files that call macros) do we report progress at the top level file only or do we compute progress for all nest levels?

tedh-shopbot, 2015-10-12:
I would be happy as a fist approximation for it to reflect current position is simple interpretation of file ... we can worry about more sophisticated things later. I don't think it works at all for openSBP at the moment.

ryansturmer, 2015-10-12:
When you say simple interpretation of the file, you mean this:

> The progress reflect the current position in the file, basically ignoring any notion of looping. So the progress would be simply be CurrentLocation/TotalLines - This would result (in the case of [...]

Is that correct?

One enhancement to this might be to simply refuse to decrease the progress - basically force monotonicity by never reducing the number, so, in pseudocode:

```
progress = 0
while(file_is_running):
   p = 100*(current_line/total_lines)
   if(p > progress):
      progress = p
progress = 100
```

Am I on the right track with the expected behavior?

tedh-shopbot, 2015-10-12:
My vote would be to not over-think it. 
Just stick with:  p = 100*(current_line/total_lines)
If someone has a file with loops and outside calls, they will know ... and if they really need to watch, they can view the flow of the code (when we turn it back on).
Mostly this is for the guy who starts a large sign file and wants to know how it is coming.
I *would* ignore calls to a macro ...

ryansturmer, 2015-10-12:
Roger.

ryansturmer, 2015-10-13:
Fixed in c240fd01

closed 2015-10-13 by ryansturmer