FabMo/FabMo-Engine #214: Macro issues ...

tedh-shopbot, 2015-10-11:
-The Macro for C6 turns on the spindle and then the file stops. I forgot to get the log and just fixed this for my work with an SO,1,1.
-We need a Macro for C9 for current posts to work ... don't know about whether we want to get involved in this MTC stuff yet ... it will be complex.
-Macro editor, like the editor, problematic on mobile especially not being able to get to the bottom of a file.

ryansturmer, 2015-10-11:
There is no macro editor - It's just the editor - so if we clear up issues with the editor, we clear them up for editing macros too.

What is the intended behavior for C9?  (Will open an issue for that)

tedh-shopbot, 2015-10-11:
C9 is tool change ... that is probably more than an issue at this point ...

ryansturmer, 2015-10-11:
We can stub in a tool change macro so that files posted with ATC output don't cause the system to choke.  What do you do with C9 currently on a tool that doesn't have an ATC?

ryansturmer, 2015-10-11:
And of course now to get through files that have a C9 you can create a C9 macro and stub in whatever you want in the macro manager.

tedh-shopbot, 2015-10-12:
Acutally it is an ATC/MTC system. The MTC version manages a manual tool change which we can't do with G2 at the moment (or at least not the way Alden wants to do it) ...

ryansturmer, 2015-10-12:
You should be able to do MTC the shopbot language, right?  Can you detail the process here, and we can discuss how we can implement it witth macros?

ryansturmer, 2015-10-13:
The buggy behavior of macro execution is fixed.  Opening another ticket for creating new macros.

closed 2015-10-13 by ryansturmer