FabMo/FabMo-Engine #216: Hand wheel occasionally locks ...

tedh-shopbot, 2015-10-11:
It is easily cleared by clicking again ... but it is problematic that it can get stuck on ...

ryansturmer, 2015-10-11:
What is meant exactly by stuck on?  Is it:

* Tool moves although wheel is not spinning
* Wheel moves even though user is not actually turning it
* Wheel retains focus when it shouldn't?
* Wheel stays stuck in the "accepting user input for turning" state even though the user has clicked away?
* System remains in "Manual" state although wheel has definitely been disengaged.

Questions:
* What is the appearance of the wheel colorization when it happens?  (Does X stay lit up if you're spinning X?)
* Are there steps to reproduce?  (I know it's tricky with this type of UI element, but if you have any hints...)

tedh-shopbot, 2015-10-11:
After spinning a bit, user removes finger -- tool continues to move.
Handwheel is still highlighted and touching it will stop the tool -- then all is normal.
That is, there seems to be some state where without continuing input, tool continues to move (handwheel image is static at this point, just highlighted)

ryansturmer, 2015-10-11:
Does the tool move *forever* in this instance, or just for a longer-than-desired distance?

tedh-shopbot, 2015-10-12:
It moves forever ... I just noticed that it happens on a pc whenever the cursor is left outside the red panel. I haven't had a chance to try on another device ...

tedh-shopbot, 2015-10-12:
Does not happen mobile, but easy to produce on my PC.

On both devices I get the following ... which I'm not sure is intended or not. If I do a lot of spinning and stop, if I remove my finger mouse, tools stops. If I keep holding finger or mouse button down but make no move, the whole accumulated spin executes ....

ryansturmer, 2015-10-12:
That is the intended behavior, but not necessarily the correct one.  I think perhaps if the wheel stops it should stop the tool, yeah?  Until we get proper feedrate scaling, etc in place...

commit b4ec00a9 (2015-10-12): Fix #216
closed 2015-10-12 by ryansturmer