FabMo/FabMo-Engine #217: Lots of little distractions with Editor on Mobile ...

tedh-shopbot, 2015-10-11:
scrolll, positioning of cursor, not typing right, block of bottom of window with bottom bar ...

ryansturmer, 2015-10-11:
Can we make a specific list?

* Scroll (Expected vs actual behavior?) 
* Positioning of cursor (Expected vs actual behavior?)
* Bottom of window is blocked by bottom bar - This is probably a separate issue - bottom bar should not overlap apps?

tedh-shopbot, 2015-10-11:
Lots of interacting issues with editing on *mobile* ... I'm thinking best to have a look and pow-wow ... but first issue is that when trying to enter a character, it frequently is locked in a mode where it just writes over the last character ...

ryansturmer, 2015-10-11:
I have seen the issue of the overwriting character on desktop as well...  See #218

ryansturmer, 2015-10-11:
I have pushed a change with a lot of cleanup to the way the ACE editor works on mobile (84b9779c11e4d3755f5f3343bee4533bab38c635) - Give it a try and see how many issues persist.

I am opening a separate ticket for the bottom bar.

ryansturmer, 2015-10-11:
Keeping open for discussion - will close when issues are confirmed as resolved.

closed 2016-05-06 by ryansturmer