FabMo/FabMo-Engine #220: OpenSBP Language fail on simple file

ryansturmer, 2015-10-12:
Some machines fail on line 35 of the attached file.

[motor_end2.txt](https://github.com/FabMo/FabMo-Engine/files/11492/motor_end2.txt)

[motor_end2_log.txt](https://github.com/FabMo/FabMo-Engine/files/11493/motor_end2_log.txt)

ryansturmer, 2015-10-12:
@tedh-shopbot I will want to look at this machine, Ted.  Neither my laptop nor my edison reproduces it, so I'm expecting it's specific to your settings (jerk, speed, etc)  - It's a G2 panic, so either the engine gave it some input it choked on that it shouldn't have, or it's crapping out due to sensible inputs that it handled poorly.  I have looked at the G-Code for the file, and (on my engine anyway) it seems OK.

ryansturmer, 2015-10-12:
@tedh-shopbot Can you supply the version of the engine you were running?  (http://tool-url:9876/version)

tedh-shopbot, 2015-10-12:
9c8506f9e25f2d4767030a8e88c091bf6ccdaa5b
I'm not going to want to type that number again ...

tedh-shopbot, 2015-10-12:
I don't get it ... git just erased the huge number I just typed ...
Anyway, I have the FabMo unit in my pocket and will have it with me tomorrow.

ryansturmer, 2015-10-12:
The number is there.  Thank you for posting.  You actually only need the first 8 digits or so.

ryansturmer, 2015-10-12:
This must be something that is misbehaving with your specific combination of settings (speed or jerk or unit values I'm guessing) since I don't see it on other tools (even when I run the same engine version as yours) We can examine the actual unit tomorrow.

ryansturmer, 2015-10-13:
This issue is with G2

https://github.com/synthetos/g2/issues/104

closed 2015-10-13 by ryansturmer