FabMo/FabMo-Engine #221: Go to position in dashboard DRO

ryansturmer, 2015-10-12:
Clicking in the position display area should launch some kind of dialog or interface for directly driving the tool to a manually entered position.

gordonab, 2015-10-13:
Is it possible to make this work similar to the way it currently works in
SB3? The current behavior only works in the keypad coordinate display. It
works by clicking on the coordinate to be changed, changing the value and
clicking the GOTO button in the keypad window. It is also possible to
change as many coordinates as you want before clicking GOTO. RIght now
there isn't a GOTO button in the DRO so some other method of sending the
tool to the new coordinates needs to be established.

ryansturmer, 2015-10-16:
Here is a state diagram of what the interaction might look like:  

[image]

Topics of discussion:
* What happens if the tool is set into motion when you're in the "Editing" state?
* Is there a way to "cancel" entered values? (Without having to go to them using the goto button)
* What are the colors for each state and where does the goto button go and is the goto (/cancel?) button visible all the time, or only after you've made a change to the position displays?

wholetthedogsout, 2016-01-18:
Woohooo!

closed 2016-01-18 by wholetthedogsout