FabMo/FabMo-Engine #227: Default macros should be specified for backwards compatibility with OpenSBP posts

ryansturmer, 2015-10-13:
C2 - Z Zero plate
C3 - Home X,Y,Z
C6 - Spindle on
C7 - Spindle off
C9 - Tool Change

gordonab, 2015-10-15:
Where did we leave off with system variables? The current custom cuts are heavily reliant on them. I'll either need to change them to use local variables to the custom cuts or test the system variables.

ryansturmer, 2015-10-15:
Some system variables are available (such as the ones for position, etc) but any "configuration" type values, such as the preferred z pullup etc are not available.  For these, I would just code in some sensible defaults directly into the macros for now.  We can discuss the future of these types of variables later.  The objective of this ticket is really just to get things stubbed in so that posts from the current tool don't fail.

ryansturmer, 2015-10-17:
I think these exist in one form or another, but could probably use some cleanup, more testing, "variableification" (more settings aware) and a license on the file.  (What are we doing for file licenses?  Is it copyleft like I've seen some of the other example files?  Or is it apache 2.0 - @tedh-shopbot ?)

tedh-shopbot, 2015-10-17:
For anything ShopBot or Handibot or general FabMo (of our creation), license should be apache 2.0 ... and not copyleft. If there is something that is a personal contribution or someone else's, etc ... then license is at their option.

tedh-shopbot, 2015-10-17:
{{In the early spirit of open source, Bill used a creative commons copyleft license on some SBP files ... basically to let people know they were public and could be utilized (this may be the earlier licenses you are referring to). Since we are the owners of all of that material, we can re-issue with apache2 ... but the copyright date on the new version would start with the re-issue date.}}

ryansturmer, 2015-12-01:
@gordonab There is also C90 - what's that one all about?

closed 2016-04-22 by ryansturmer