FabMo/FabMo-Engine #233: Progress bar immediately goes to full with no % number next to it in desktop mode.

gordonab, 2015-10-14:


ryansturmer, 2015-10-14:
Can you post the file you ran?

wholetthedogsout, 2015-10-15:
In the bottom bar?

gordonab, 2015-10-15:
Yes, the progress indicator in the bottom bar. The file was the sblogo.
I'll attach the file when I get to the office.

gordonab, 2015-10-15:
This is the file I ran when the progress bar went to full and didn't display the percentage.
 
[shopbot_logo.sbp.txt](https://github.com/FabMo/FabMo-Engine/files/14134/shopbot_logo.sbp.txt)

ryansturmer, 2015-10-15:
Only the bottom bar progress indicator?  Or the one in the DRO as well?

gordonab, 2015-10-15:
Both of them are showing full the next time a file is run.

ryansturmer, 2015-10-15:
Having a hard time reproducing - is this a file that's run to completion before the next one starts?  Is any progress indicated during the run?  Or is it just stuck at 100% - Also what is your engine version (http://tool_url/version)

gordonab, 2015-10-15:
Both progress bars immediately go to full when starting the SB Logo file as
soon as the file starts. It can be a file that is resubmitted and run to
completion or a new file.

info: G2 Firmware Build: 83.09 [engine]

So far this is the only edison I have with the new image. I will try
another one shortly and see if it behaves the same way.

ryansturmer, 2015-10-15:
Sorry Gordon, not the firmware version, but the engine version:  Visit http://192.168.1.x/version - With your IP subbed in

gordonab, 2015-10-15:
Tried this, but get "The page at /version was not found."

ryansturmer, 2015-10-15:
Sorry, did you include the :9876 ?  If you don't have that page, you're probably using a pretty dated version of the engine somehow, which is worrisome if it's a new image.

gordonab, 2015-10-15:
From 192.168.1.200:9876/version :

{"status":"success","data":{"version":""}}

gordonab, 2015-10-15:
I just imaged another and it had the same empty version

ryansturmer, 2015-10-16:
Hmm, that's not necessarily am old version it might be related to the git issue you saw the other day

ryansturmer, 2015-10-16:
Yep, it was an old version on the image that was non-updatable.  This has been resolved in some recent commits.

closed 2015-10-16 by ryansturmer