FabMo/FabMo-Engine #237: Trouble getting on shopbot-private network

tedh-shopbot, 2015-10-21:
Noticed today that I am unable to get several fabmo devices on the shopbot-private network. Used --wifi and this seemed to work with 'wpa_cli status' showing both were on the network and giving an IP. But, devices did not show up on network. When they were rebooted that showed on FabMoNet.

Any possible relation to the problem Worden reported related to bug in edison and the 3'rd octet number not working unless it was a .1  ? In this case we were at .33  ??

Or perhaps different security or more complexity on the shopbot network?

ryansturmer, 2015-10-21:
When you say "not showing up" on shopbot network, what do you mean exactly?  What is the output of 'ifconfig' in this instance?

ryansturmer, 2015-10-21:
So, as an example, you say wpa_cli shows an IP address, but I assume when you visit http://ip_address:9876 you're getting a host not found, or a connection refused?

ryansturmer, 2015-10-21:
also on these clients can you "see out?"  (ping google.com, for instance)

tedh-shopbot, 2015-10-21:
have not explored much further ... yet and forgot to try the ifconfig while there since I had assumed the wpa_cli info that it was on the network was about the same thing. And strangely, fing shows something there with the wrong name but correct mfg ... ?

ryansturmer, 2015-10-22:
Wpacli has the info too, the output of either works be acceptable to aid diagnosis

gordonab, 2015-10-22:
I tried you problem Edison today and was able to successfully run --setup
and got through the Wi-Fi to fabmonet. It gave a IP address that I couldn't
see with fing. I brought it home to try on my home network.

tedh-shopbot, 2015-10-23:
Cool … I’ll be back on FabMo tomorrow … =t

gordonab, 2015-10-23:
So, this one is confusing. I have been successful at getting a FabMo to run over the shopbot-private network. It has take 2 attempts to get it there. Here's how it went:

1. use configure_edison --wifi to connect to the shopbot-private network and was given the IP address 192.168.33.106

2. connected using putty via the console line (com) without any issues

3. connected using putty via the IP address without isssues

4. tried to bring the dashboard up using Chrome. The dashboard started loading, showed the top and side frames, but displayed the rotating wait graphic without stopping.

5. shutdown the Edison through putty and restarted. Now, I could still get in through the com line, but after recycling the power it shows the original FabMoNet IP address. I tried switching my computer to FabMoNet to bring up the dashboard, but it could find it.

6. using configure_edison --wifi I connected the Edison back to FabMoNet and was able to run the dashboard as we usually do.

7. switched the Edison, my computer and phone back to shopbot-private and am able to bring up the dashboard in Chrome and was able to install the apps. The funny part here is that I can see the IP address (192.168.33.106) in Fing, but the SSID is blank. Also, Jimmy's tool doesn't see it.

Here's the log file from the time it was changed to shopbot-private to right before the first move of the handiwheel.

Oct 23 17:25:26 fabmo_ed_21 systemd-timesyncd[172]: No network connectivity, watching for changes.
Oct 23 17:25:26 fabmo_ed_21 udhcpc[641]: Sending discover...
Oct 23 17:25:26 fabmo_ed_21 wpa_cli[276]: Sending discover...
Oct 23 17:25:26 fabmo_ed_21 udhcpc[641]: Sending select for 192.168.33.106...
Oct 23 17:25:26 fabmo_ed_21 wpa_cli[276]: Sending select for 192.168.33.106...
Oct 23 17:25:26 fabmo_ed_21 udhcpc[641]: Lease of 192.168.33.106 obtained, lease time 172800
Oct 23 17:25:26 fabmo_ed_21 wpa_cli[276]: Lease of 192.168.33.106 obtained, lease time 172800
Oct 23 17:25:26 fabmo_ed_21 systemd-timesyncd[172]: Network configuration changed, trying to establish c                             onnection.
Oct 23 17:25:26 fabmo_ed_21 systemd-timesyncd[172]: Using NTP server 216.239.36.15:123 (time3.google.com                             ).
Oct 23 17:25:26 fabmo_ed_21 systemd-timesyncd[172]: Using NTP server [2001:4860:4802:36::f]:123 (time3.g                             oogle.com).
Oct 23 17:25:26 fabmo_ed_21 systemd-timesyncd[172]: Using NTP server 216.239.38.15:123 (time4.google.com                             ).
Oct 23 17:25:26 fabmo_ed_21 wpa_cli[276]: /etc/udhcpc.d/50default: Adding DNS 192.168.33.9
Oct 23 17:25:26 fabmo_ed_21 wpa_cli[276]: /etc/udhcpc.d/50default: Adding DNS 192.168.33.5
Oct 23 17:25:26 fabmo_ed_21 systemd-timesyncd[172]: Using NTP server 216.239.32.15:123 (time1.google.com                             ).
Oct 23 17:25:26 fabmo_ed_21 systemd-timesyncd[172]: interval/delta/delay/jitter/drift 64s/+0.041s/0.054s                             /0.038s/+240ppm (ignored)
Oct 23 17:25:57 fabmo_ed_21 systemd[1]: Starting OpenSSH Per-Connection Daemon (192.168.33.105:65517)...
Oct 23 17:25:57 fabmo_ed_21 systemd[1]: Failed to reset devices.list on /system.slice: Invalid argument
Oct 23 17:25:57 fabmo_ed_21 systemd[1]: Started OpenSSH Per-Connection Daemon (192.168.33.105:65517).
Oct 23 17:25:57 fabmo_ed_21 systemd[1]: Couldn't stat device /dev/pts/ptmx
Oct 23 17:26:05 fabmo_ed_21 sshd[656]: Accepted password for root from 192.168.33.105 port 65517 ssh2
Oct 23 17:26:05 fabmo_ed_21 sshd[658]: lastlog_openseek: Couldn't stat /var/log/lastlog: No such file or                              directory
Oct 23 17:26:05 fabmo_ed_21 sshd[658]: lastlog_openseek: Couldn't stat /var/log/lastlog: No such file or                              directory
Oct 23 17:26:30 fabmo_ed_21 systemd-timesyncd[172]: interval/delta/delay/jitter/drift 128s/-0.018s/0.020s/0.047s/+168ppm
Oct 23 17:28:29 fabmo_ed_21 node[192]: debug: Setting the time to Fri, 23 Oct 2015 17:28:29 GMT [engine]
Oct 23 17:28:29 fabmo_ed_21 node[192]: debug: Executing shell command: timedatectl set-time 2015-9-5 17:28:29; timedatectl [util]
Oct 23 17:28:29 fabmo_ed_21 dbus[214]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service'
Oct 23 17:28:29 fabmo_ed_21 dbus-daemon[214]: dbus[214]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service'
Oct 23 17:28:29 fabmo_ed_21 systemd[1]: Starting Time & Date Service...
Oct 23 17:28:29 fabmo_ed_21 node[192]: debug: client connected [websocket]
Oct 23 17:28:29 fabmo_ed_21 dbus[214]: [system] Successfully activated service 'org.freedesktop.timedate1'
Oct 23 17:28:29 fabmo_ed_21 dbus-daemon[214]: dbus[214]: [system] Successfully activated service 'org.freedesktop.timedate1'
Oct 23 17:28:29 fabmo_ed_21 systemd[1]: Started Time & Date Service.
Oct 23 17:28:29 fabmo_ed_21 node[192]: Local time: Fri 2015-10-23 17:28:29 UTC
Oct 23 17:28:29 fabmo_ed_21 node[192]: Universal time: Fri 2015-10-23 17:28:29 UTC
Oct 23 17:28:29 fabmo_ed_21 node[192]: RTC time: Fri 2015-10-23 17:28:29
Oct 23 17:28:29 fabmo_ed_21 node[192]: Time zone: Universal (UTC, +0000)
Oct 23 17:28:29 fabmo_ed_21 node[192]: NTP enabled: yes
Oct 23 17:28:29 fabmo_ed_21 node[192]: NTP synchronized: yes
Oct 23 17:28:29 fabmo_ed_21 node[192]: RTC in local TZ: no
Oct 23 17:28:29 fabmo_ed_21 node[192]: DST active: n/a
Oct 23 17:28:38 fabmo_ed_21 systemd-timesyncd[172]: interval/delta/delay/jitter/drift 256s/-0.014s/0.033s/0.046s/+140ppm
Oct 23 17:28:39 fabmo_ed_21 node[192]: info: Got a machine state change: manual [machine]
...

ryansturmer, 2015-10-29:
The more interesting log to collect is probably the one right before the *first* attempt to get on shopbot-private.  

configure_edison does a bad job of managing wireless networks, and it's unsurprising to me that it forgets your selected network so quickly.  Hopefully the new networking scripts will fix this.

ryansturmer, 2015-11-04:
New network management has no difficulty so I'm closing this.  There are still auto-discover problems, but it is a separate issue.

closed 2015-11-04 by ryansturmer