FabMo/FabMo-Engine #240: Job Manager History view ... mods?

tedh-shopbot, 2015-10-24:
I think for me, the display might be better if the file duration were added to the main display along with how long ago ... and all the stuff over in the right hand "..." were moved to the "+" button. It seems to me the important stuff and is often hard to find way over on the right ...

ryansturmer, 2015-10-25:
The thing about loading up the columns is that on mobile they get scrunched, and then they're basically useless - what columns would you want to be the first to disappear to save space when things are reduced.

I think the impulse to have the menu actions under the "+" comes from using the manager for so long with them there.  I never should have put them there - they belong in an action menu, that was just the easiest place to put them.  I think mechanics-wise the action menu on the right side makes the most sense, and if anything, we should ditch the "+" altogether, and maybe put a "more detail" in the right side menu.  We should definitely collect opinions from people who have not been using the system till now.  I habitually go to the "+" for extended job actions, but I think this is habit due to having had them there for so long, not an impulse based on the way things "should" work.

tedh-shopbot, 2015-10-25:
I don't know ... but for me, I like having them in the + on the left. That is where I expect to go to dig down and it never gets accidentally hidden. If you totally remove the right, 4th, column (...) putting those options in the + then that column can be used for the file duration info and you will not have increased the size of the spread. On my phone, those 4 columns would work ok.

wholetthedogsout, 2015-10-26:
Im working on making the table view look cleaner on mobile. I think we can probably keep the 4 columns, we could switch the position of the (+) and the (...). Generally the higher "clicked" objects go on the left.

ryansturmer, 2015-10-27:
It's probably worth thinking about the "verbs" in the system.  (...) definitely means "menu" but it can also mean "more information" - so there's that to think about.  (+) Can mean "more information" but it can also mean "add a thing" We want to do what people are expecting.  I think switching the position of the two is a good start, but the plus on the right side feels weird to me... maybe the behavior of the plus might be achieved by a click on the row outside the (...) - or something.  Let's start by switching places and see how it feels.  There's also the option of not doing row expansion, and having a click take you to a "job info" page with all the relevant deets.

tedh-shopbot, 2015-10-27:
So I was suggesting eliminating the one (the +) that only had the added info of the duration and add it to the data across the top (the other item is redundant) ... there is no need for this button, especially having it so prominent. 

I have been wondering what you mean by "+" ... now I see! It turns out that on my devices the plus symbol is so small in the green field that I can not see it without careful inspection. Thus, the verb I have been seeing is "green button" and for me the "green button" has meant "expose the important additional opportunities"

The "..." verb is currently invisible and difficult to discover on the far right side where it is hidden unless you are in a full screen up. So on most devices, the user does not know there is an expansion or additional option, even though it is the primary feature here.

My vote is to thus simplify to one addition effector ... however it is symbolized, and put it where it is visible and natural to explore on the left ...

So at the moment, there are 5 columns in the table: (+button) - name - status - run at - ("...")
I suggest keeping the same number of columns:  (a verb) - name - run at - duration - status  

I personally find the green button nice and visible and certainly expect it to lead to more options. So I would leave it where it is. But I don't really care whether it is changed to some other verb, but one expansion button is enough. We are already using hamburger and "MENU" to mean menu ... seems like adding a third is a complication ... thus logically, the green button might be a hamburger :)

ryansturmer, 2015-10-27:
Nix row expansion, move action menu to the left, change columns orders to what ted has described - sounds good to me.

Let's keep the (...) as the verb for action menu for now.  The green isn't thematically appropriate, we can adjust colors to improve visibility if necessary.

wholetthedogsout, 2015-10-27:
Sounds good

closed 2015-10-28 by wholetthedogsout