FabMo/FabMo-Engine #245: Bottom Bar Improvements

ryansturmer, 2015-10-30:
I think the following improvements should be made to the bottom bar:

1) No pull-tab - not necessary, and covers important features in apps sometimes
2) Bottom bar should not cover the rest of the interface
3) bottom bar should not be dismissable while a job is running, and should dismiss itself after job completion

wholetthedogsout, 2015-10-30:
Agree with 1 and 3, but a little confused on 2. Wouldn't you want the bar very visible while a job is running?

ryansturmer, 2015-10-30:
Ah yes. 

Bottom bar should be persistent, but "shove" the interface out of the way - the problem is when the bar is up, the dash/app client area doesn't resize, so the bar is covering the bottom of the app, which is often where controls are.  I'm imagining when the bar is up, it scoots the dash out of the way, in the same way that the DRO does when it's on desktop size.

ryansturmer, 2015-10-30:
Also, we still have this jazz going on:

[image]

wholetthedogsout, 2015-10-30:
Yeah that will go away because there will be no next job. And I will add padding to the bottom when the bar comes up so it will push the space up too. That makes sense.

ryansturmer, 2015-10-30:
Oh yeah of course :+1:

wholetthedogsout, 2015-10-30:
off topic but since you are on here can you give me access to your folder where the edison image is?

closed 2015-12-07 by wholetthedogsout