FabMo/FabMo-Engine #246: An easy to read FabMo version number for users ...

tedh-shopbot, 2015-10-31:


ryansturmer, 2015-11-01:
This implies that we have a version number, and that implies that we have a release schedule that we sorely need!

I like semantic versioning http://www.semver.org/

TLDR:  It's major.minor.patch where major means you broke the API, minor means you made a backwards compatible feature improvement, and patch means you made a backwards compatible bugfix.  When you change a higher order component of the version number, the lower order ones go to zero.

There are almost as many strategies for doing releases as there are software projects.  I always like to err on the side of "keep it simple"  - The model described here: 

http://nvie.com/posts/a-successful-git-branching-model/

Is pretty good.  The master-as-release branch thing is a little troublesome with a team that is new to Git, mainly because you get a lot of people pushing to the master branch inadvertently because they're still learning the tools, and if the master is the release branch, you end up with a lot of commits that you don't want there, so I would consider revising this strategy to make the master branch the development branch, and have a permanent "release" branch which behaves like the master in the system described above.

I invite the whole team to comment on this one.  This isn't something to be taken lightly, and you want to choose a development strategy that fits culturally, and is maintainable as the team and the project grow.  What do you guys think?  @jlucidar @wholetthedogsout @tedh-shopbot @eliwilliams @gordonab @Alex-Canales (please tag anyone I left out in your replies)

ryansturmer, 2015-11-01:
It seems like a digression, but this really is meant to address the issue on the ticket, namely, how do we set up a version number that means something for users.  I think from a technical perspective, the git hash is the most informative information we can harvest from a build, because it describes the code completely no matter where it came from (unless it was locally modified), and in an open source environment, where you're going to have a lot of people running the "edge" version of the software, we're going to want that in any case, but users don't interact with a big long hash very well, and if it's important to us for them to have a version number that they identify with, this is a good way to do it.

tedh-shopbot, 2015-11-01:
I like the idea of a "release" branch ... particularly if we want to use that git site for one button (or auto) updates. Similarly, my interest in version numbers is so that -- particularly for the case of interactions with a normal tool user -- it is very easy to identify for us (tech support person) and the user what version they are on. For this, the semantic versioning is fine and is what ShopBot has always done. (I don't see a problem with only versioning the release versions ...)

Now a little deeper question: Consider a Handibot vs a DIWire -- does each tool provider maintain their own "release" version in order to cover the specifics of their tool? 
     - In the best of all worlds, we will be able to specify configuration and settings files that are separate from the tool, and a user could basically just update the engine code from a common source, while going to a provider for configurations??
     - In such a world, you might want to allow 2 updates ... one for the engine and one for the configuration; you might also want a way to save specialized configurations and settings against loss and/or be able to optionally select configuration files ???

jlucidar, 2015-11-01:
I agree on the versioning format, I already use it for my projects and it's the best one I know. I also agree that we should adapt our branching, and I like the one Sturmer presented. We sort of already using it for new features and it seems pretty clean to me.

ryansturmer, 2015-11-02:
I have also opened #248 which is an issue you have mentioned ted, and is inextricably tied to this discussion.  FWIW, I think perhaps 2 version numbers is appropriate, one for the engine itself, and one for the configuration and its system apps.  Technically, the system apps are their own sort of projects, and so each might have a version, which you could collate into a "version report" that shows the engine, configuration, and instance versions of all installed apps.  Might be overkill, but it depends on our objective.

tedh-shopbot, 2015-11-02:
I double we'll ever be able to control configuration versions or app versions ... but it might be useful if we had in place a mechanism to at least report them if available.

ryansturmer, 2015-11-02:
We can't control them, but we can require that they be reported, that way, ther'es something to go on, and if implementors don't follow sensible versioning practices, we can at least implicate them for it, rather than having to shoulder the blame ourselves

commit 16f21412 (2016-02-08): Update app reload rules - add version number to config app - close #246
closed 2016-02-08 by ryansturmer