FabMo/FabMo-Engine #251: In Network Manager: need double click equivalent on Android

tedh-shopbot, 2015-11-06:
New network management logged in well on my home network from PC.
But could not initiate a transfer from Android (no double tap) or maybe I'm missing something ... 

Also, first time after rebooting, it did not come up in AP until I hit AP button ...

ryansturmer, 2015-11-06:
Sorry about the double tap - that's vestige from the original network app which we never tested on mobile.  I imagine we want to add a (...) menu like we have in the job manager app, with options like 'Connect to this network, Forget network, etc."

Had you ever used this edison on your home network before?  If the tool would actually come up on your network the first time if it had ever seen your network before.  Also, it takes at least one re-run of the engine to replace the wpa_supplicant file, so first-time weirdness is expected, but only after the initial installation.

tedh-shopbot, 2015-11-06:
Yes ... first option. It was already on the network when I put the new stuff on. So yes ... probably just went back on and I hadn't looked.

ryansturmer, 2016-01-21:
Single tap/click to join a network now, with some other improvements.

closed 2016-01-21 by ryansturmer