FabMo/FabMo-Engine #252: Problem with IF in sbp file (macro)

tedh-shopbot, 2015-11-07:
My Line ...
IF (&last_bend_direction = 1) THEN GOTO Position   ' Don't duck unless we have to
{{just skips line; Position: label looks correct further down}}

My error ...
warn: Unhandled Command: {"type":"cmd","cmd":"IF","args":["(&last_bend_direction = 1) THEN GOTO Position"],"comment":[" Don","t duck unless we have to"]} [sbp]

ryansturmer, 2015-11-08:
This is an IF getting mistakenly parsed as a command (since it's a 2 character mnemonic) - this is a regression - it didn't used to do it.  Probably a precedence issue in the parser.

ryansturmer, 2015-11-08:
@tedh-shopbot  Only partly!  If you remove the parentheses in your IF test, it will work.  The parser doesn't recognize parentheses here.  I am working to fix it.  But you can just remove the parens in the short term.

tedh-shopbot, 2015-11-08:
got it

ryansturmer, 2015-11-08:
To be clear, you should be able to use parentheses generally in your IF test - you just can't enclose the whole conditional in parentheses.

commit 1060f89f (2015-11-08): Fix #252
closed 2015-11-08 by ryansturmer