FabMo/FabMo-Engine #254: Not running next command after macro call in file

tedh-shopbot, 2015-11-09:
This only runs the first macro (and does not record the variable):
	    &next_feed = 1.5
	   _a,FEED
	    &next_bend = 35
	   _a,BEND

This only runs the macro
    C#, 33
    MX, 45

ryansturmer, 2015-11-09:
Please provide the contents of your macros

wlyoung, 2015-11-14:
I think I'm seeing the same issue. If I run a file that starts with C6 as
the first line and then Move commands like this:

C6
JZ,0.500000
J3,2.250000,0.585000,0.236200
M3,2.250000,0.585000,0.000000
M3,2.245956,0.585126,-0.001288
M3,2.241830,0.585515,-0.002607

It runs the macro and then shuts down. If I replace the macro with the SO
command like this:

SO,1,1
PAUSE 3
JZ,0.500000
J3,2.250000,0.585000,0.236200
M3,2.250000,0.585000,0.000000
M3,2.245956,0.585126,-0.001288
M3,2.241830,0.585515,-0.002607

It runs fine.

ryansturmer, 2015-11-14:
Bill, this has been fixed to my knowledge, can you please report the version of the engine you're running?  (Go to http://yourip:9876/version and paste the result there.

wlyoung, 2015-11-14:
{"status":"success","data":{"version":"599d2503108955cd69e6ae96861251cfb2d27454"}}

ryansturmer, 2015-11-14:
Thank you for the diagnostic information.  Your version has all of the fixes I made for this problem, so it appears there are still some issues.  Can you please post the contents of your C6 macro?  Is it just the SO,1,1 line?

I appreciate the detail in reporting here, it really makes it easier to track this stuff down.  You guys are the best.

wlyoung, 2015-11-14:
It's the default #6 macro with only:

SO,1,1
PAUSE 3

in it. I haven't made any modes to it.

ryansturmer, 2015-11-17:
Is anyone else seeing this one?  My engine doesn't do it.

wlyoung, 2015-11-17:
I'll retry when I get back in the shop and see if I can get any more
enlightenment. Is there any special syntax, other than just C6?

tedh-shopbot, 2015-11-17:
Had this problem before the last fix; but have not had a chance to get back and putter with the bender ... which is where I was using a lot of (embedded) macro calls, etc ...

ryansturmer, 2015-11-17:
No, this should be completely compatible with prior versions of the shopbot language (that's the goal, anyways)

ryansturmer, 2015-11-23:
This is still happening - but it is timing specific so it's hard to track down.  Hoping to reproduce the situation soon.

ryansturmer, 2016-01-07:
Fixed AFAIK.  Will reopen if it resurfaces.

closed 2016-01-07 by ryansturmer