FabMo/FabMo-Engine #256: Fabmo doesn't see my wifi network

wlyoung, 2015-11-13:
In the WIFI network list Fabmo only shows one network that is at a local business. It doesn't list my network at all which is in the same room. Broadcast SSID is turned on and it has the same network  security setting as the one it sees.

ryansturmer, 2015-11-13:
Can you try to reboot and reconnect to the tool a few times in AP mode and see if the network list changes at all?

wlyoung, 2015-11-13:
We've rebooted everything we could think of, multiple times No love so far.

wlyoung, 2015-11-13:
My local network just appeared...not sure what changed? How do we connect
to it and input the wifi password?

ryansturmer, 2015-11-13:
Double click the network name, and input the key in the dialog.   The network appearing is luck-of-the-draw - I need to make the scans more aggressive.  Did it ever show more than one network in the list? Or did your network turn up alone?

wlyoung, 2015-11-14:
When my network finally showed up it was in addition to the other
network...they both showed at the same time.

Brian, James, and I tried clicking everything we could click yesterday but
no luck with entering the password. This morning my network showed up when
I first turned it on and I let it run while I had my first cup of coffee.
After reading your email I went back and double clicked the network name
and was able to enter the password and connect just fine. Don't know if we
just hadn't found the right clicking spot yesterday, or if it needed to run
for a while?

All seems to be working now and connected to my local network

wlyoung, 2015-11-17:
Just to put a point on this, once I got everything figured out the network
stuff has worked great. I've turned my router off and on, changed
passwords, and changed SSID. It would fall back into direct connect like it
was supposed to and then connect when the network was up.

Just as an fyi though, the double tap to get the password entry form is not
intuitive on a phone

Bill Young

mobile 757-999-5592
office  888-680-4800 ext 133

ryansturmer, 2015-12-01:
The double tap is a vestige of an older interface and will be fixed.  Closing this for now.

closed 2015-12-01 by ryansturmer