FabMo/FabMo-Engine #257: weird motion?

wlyoung, 2015-11-14:
Things are going well and we've been able to connect and run files, but at one point negative X moves became very rough. They gave the impression that something mechanical like the nut had come loose. All other axis worked beautifully. It gave the impression that there was a different step resolution in + and - X moves

We turned it off and could easily move the lead screw in both directions, but when we turned the handibot back on we still had the same problem...negative X moves were awful and missed steps where everything else was fine.

I put the old control card back in and everything worked normally, so then re-installed the Fambo card and no problems since. No insights on what happened but it was very weird.

ryansturmer, 2015-12-10:
Steps to reproduce?

wlyoung, 2015-12-10:
It has never appeared again after much cutting. I'm closing and will reopen if it happens again

closed 2015-12-10 by wlyoung