FabMo/FabMo-Engine #258: executing a file from the editor doesn't seem to work for me

wlyoung, 2015-11-16:
This may be an sbp/gcode issue, but if I type a ShopBot command like MX,6 into the editor and try to run it using any of the execute options, nothing happens. The same with the Macro editor...I can't run a macro from any of the execute menu items when I'm in the macro editor, but it will run from the list of macros if I click the Run/Play triangle dohickey.

ryansturmer, 2015-11-16:
Are you selecting the language from the language menu?

wlyoung, 2015-11-16:
Obviously not! Works now

ryansturmer, 2015-11-16:
The language selection should "stick" so you don't have to do it every time (except with the macro editor, which will get fixed)  - remember, you can Ctrl+Enter to run quickly, if you're on a PC

closed 2015-11-17 by ryansturmer