FabMo/FabMo-Engine #259: I like the gofabmo.exe network-fabmo-finder ...

tedh-shopbot, 2015-11-17:
In the short run, it seems like we are going to need this and the android version (ios maybe?) ... so I'm thinking that if it would give-up the tool name, it would be more helpful ... (or, even send a link to a browser).

Maybe a project for Jimmy ... give them all the same name, look, and feel ?

ryansturmer, 2015-11-17:
I think we need to have a discussion about tool finding in general - it doesn't work on all networks... we should discuss strategies for networks where the UDP broadcasting technique we use is forbidden, or the relevant ports are blocked, etc.

jlucidar, 2015-11-17:
Let set a skype/hangout meeting next week about this point.

jlucidar, 2015-12-24:
Will you have some time soon to talk about this ?

ryansturmer, 2016-01-07:
I'm closing this, since we have Jimmy's tool now.  We can continue discussion there.

closed 2016-01-07 by ryansturmer