FabMo/FabMo-Engine #261: Input #1 on FabMo DT is a bit flaky, flashes on and off when touching plate

gordonab, 2015-11-24:


ryansturmer, 2015-11-24:
Is this the input for the ZZ plate?

gordonab, 2015-11-24:
Yes it is

Ryan-Patterson, 2015-11-30:
I see the same type of issue on input #2.  It will show triggered when it is not.

gordonab, 2015-11-30:
I'll take a look at this in the morning.

ryansturmer, 2015-12-01:
Ryan, when in particular does it show up on 2?  Is this a when-you-touch-it kind of thing?  Or is it flaky during tool motion?  Or just at random.

Ryan-Patterson, 2015-12-01:
I see it when testing a macro to home the machine.  I ran a macro to touch the Y of a NC prox switch connected to input #2.  After running this a couple times input #2 indicates the switch is triggered.  Activating the switch does not affect the display of the switch.

gordonab, 2015-12-01:
Can you share this file or paste the text here and I'll see what I can determine on the tool.

Ryan-Patterson, 2015-12-01:
I do not have the file here.  But it is basically the same file the is on the Desktop out back.  The Z Zero macro on the machine out back also shows the same type of behavior.

ryansturmer, 2015-12-01:
@Ryan-Patterson  Do you see the switch being "flaky" on your machine at home - ie: toggling when the input is touched or at random?

Ryan-Patterson, 2015-12-01:
Yes.  I saw the issue on input 1 here and number 2 on my home machine.

ryansturmer, 2015-12-01:
This feels like a hardware issue.  We may want to examine the schematic of the inputs.  Could be we're missing some pull-ups.

ryansturmer, 2016-03-01:
Closing this as a hardware issue and not an engine problem?

closed 2016-03-01 by ryansturmer